AronHetLam/ATEM_tally_light_with_ESP8266 #1: Max amount of Tally’s

grahamrobertslx, 2020-09-05:
Seems to be an issue that if more than 5 Tally's are connected other tally's won't connect to the ATEM

grahamrobertslx, 2020-09-05:
Actually this could be because of how many connections the ATEM will accept

AronHetLam, 2020-09-06:
It's a limitation by the Atem. But if you look at the beta release I made the lights into a server as well, so that you can connect one to the switcher, and the rest to the first device. (You only need to update the main device.)

grahamrobertslx, 2020-09-06:
I have 8 made, I'll try the beta today and let you know how it works.

grahamrobertslx, 2020-09-06:
So just been testing it adding one by one to the first tally and it works well. Got to Tally 7 and it wouldn't initialise to the Node Tally. Work round was to use the IP of the desk on Tally 7 and use that as a second node to do tally 8. So I'd probably limit 6 tally's to a Node. All very good though!! Not really any latency issues. Good Work!!!

grahamrobertslx, 2020-09-06:
Apologies limit to 5 Tally's per node

AronHetLam, 2020-09-06:
Interesting. In the code I set a limit of 20 - but it just might not be able to handle more than 5 then. - I still need to test this, which is why it's still in beta.
You could also daisy chain them if you need the ATEM's connections for software control.

grahamrobertslx, 2020-09-07:
With the 5 connected to the 1 node it is very stable and didn't miss a beat once. Very good work!

AronHetLam, 2020-09-10:
It's now in included in release 2.0

closed 2020-09-10 by AronHetLam