AronHetLam/ATEM_tally_light_with_ESP8266 #3: Latest Atem firmware support?

andrewhearn, 2020-09-13:
Hi, 

This looks great -I'm currently looking to move from vmix to using an Atem Mini Pro - and am building a tally system...

https://www.skaarhoj.com/fileadmin/BMDPROTOCOL.html says that their libs only support Atem firmware befreo 7.5.0 - the current version looks to be 8.4 according to Blackmagic...

Does the tally s/w (and libs) still work with the current firmware versions? Thanks!

- Once I have the Atem, I'll be looking to modify the code slightly to use a NeoPixel as per this project: https://github.com/gjnijenhuis/Arduino-vMix-NeoPixel-tally

AronHetLam, 2020-09-13:
The libraries from skaarhoj still work, they just don't support the features introduced by ATEM after that version. For tally with the minis there's no issues.

If you get the version in this repo I added support for getting streaming status myself. That's eg something that's more recent than Skaarhoj had made public.

closed 2020-09-13 by AronHetLam

andrewhearn, 2020-09-13:
That's great! Thank you.