AronHetLam/ATEM_tally_light_with_ESP8266 #4: FastLED support

andrewhearn, 2020-09-17:
I see you added a 2nd LED - that's great. I'm looking at using an Adafruit NeoPixel and using the FastLED library to drive it - all easy to add into your code - except I'll probably be adding it in a way that won't be up to your coding standards :-)

It would be great if you added support as a third LED using the FastLED library - but  understand you may not want to - that's fine :-)

I'm hoping to use these at our live streaming services starting next week.

Andrew

jhollowell41, 2020-09-17:
Could you explain the purpose and how the 3rd level unit functions?  Sorry if I overlooked the explanation on github. 

Jody

AronHetLam, 2020-09-17:
We have two different things mixed up here @jhollowell41, I'll address @andrewhearn first. :)

A second LED will probably be one of the last things I'll implement in the releases for the D1 mini version (I still need to figure out how i want to implement it in the settings). 
My goal is to keep it simple and without need of coding, for those out there not comfortable with that. For those who are more comfortable with electronics, code and Arduino, such as you, It's just as easy you add the fast LED features yourself, with the specific setup you have. It's nice to see people customize the project - I already saw someone make an on air sign with an LED strip. 
- I hope it makes sense why I probably won't add it. Sometime I might add a sketch for the m5stick-c, but we'll see..

@jhollowell41, did you see my guide in the wiki? There's a diagram showing the connection path and IP addresses of an example setup.
The 3rd level unit (as I might have called it poorly) refers to how the unit is connected to the swithcer/other tally units. The ATEM switchers only allow for 5-8 simultaneous connections, which is too little if you have eg. 10 cameras you want tally lights for. To solve this you can connect a unit to another unit that's already connected to the switcher (as if that unit is the switcher). The next problem then is that the D1 mini can only handle 5 other units connecting to it, so you need to daisy chain them if you need that many units.
- I call these units 2nd level units, as they are connected one step away from the switcher itself.
The next problem then is that the D1 mini can only handle 5 other units connecting to it, so you need to daisy chain them if you need that many units.
- What I call 3rd level units are units connected to a 2nd level unit. 

 I hope that made somewhat sense.

AronHetLam, 2020-09-19:
@jhollowell41
I'll close this issue, as the fastLED question is answered. If you're having any issues feel free to post a new issue here 🙂

closed 2020-09-19 by AronHetLam