AronHetLam/ATEM_tally_light_with_ESP8266 #5: Customize Light #2 independently from Light #1 Request

jhollowell41, 2020-09-28:
More of a request rather than an issue.  Would be nice to be able to customize the program function of Light #1 separate from Light #2.  Light#1 for camera person and Light #2 for stage.  Light #2 (Stage light) to be only red when camera is live, otherwise off.  Light #1 (in our case) to be green at all times unless live (red).  This is a great project!!

AronHetLam, 2020-09-28:
It's a work in progress, but basically it's already there with slight modifications to the loop function. I don't know how much you know about coding an Arduino, but you might be able to hard code the modification yourself (There is some commented code you can use in there).
The thing is that I need to implement it into the website and update the guide and everything at once, or else it'll be confusing. 😉

AronHetLam, 2020-09-28:
The code is there now,s o that you can set it up through the website - documentation will come later this week.

closed 2020-09-28 by AronHetLam