AronHetLam/ATEM_tally_light_with_ESP8266 #6: Firmware ATEM

gliepins, 2020-10-20:
Dear friend,
I am now testing this setup as you described. However Im getting no connection to the switcher. Before I dig deeper, I found that Skaarhoi librarys only support models firmware below 7.5. Current one is 8.5.
https://www.skaarhoj.com/fileadmin/BMDPROTOCOL.html (red part)
DO you have any comment on that?

Thanks!

AronHetLam, 2020-10-20:
I know, but it's not an issue.
The libraries just don't support features newer that 7.5.
Tally still works, and I'm pretty sure everything else documented on his site is still functional. I made this project after that was the case 🙂

I manually added the ability to see whether it's streaming or not, so if you're missing something you can add it manually if you know how to.

AronHetLam, 2020-10-20:
It sounds like you managed to program it. Probably you have an issue in your network setup.
Are you able to connect to the switcher with software control over WiFi? The tally lights connect the same way as software control does.

gliepins, 2020-10-20:
Alright.. Then we must find the problem. I connected everything, all sides are pinging, but it says : Disconnected - No response from switcher.

AronHetLam, 2020-10-20:
I don't have enough information about your setup.
Maybe a screenshot of the setup page and the devices' IP addresses can help me find the problem.

Also did you try to connect to the switcher with software control over WiFi?

gliepins, 2020-10-20:
I will later tonight post pics and setup. I really want to get this thing going! 
But now - what do you mean by software control?

AronHetLam, 2020-10-20:
Atem software control. The program that lets control you Atem switcher from your pc.

gliepins, 2020-10-20:
/G

gliepins, 2020-10-20:
I will later tonight. Yes, i opened control center. All good with ip addresses. All sides are pingng eachother.

AronHetLam, 2020-10-20:
What do you mean with all sides are pinging each other?

gliepins, 2020-10-20:
Means that network is working. 
Atem switch up
Arduino up
Pc with Atem soft up

gliepins, 2020-10-20:
Not arduino but esp8266 chip with led. Sory

AronHetLam, 2020-10-20:
Ok, must be a network configuration problem then. If it's on the same network and has the correct switcher IP address it should connect automatically.
Sometimes routers makes wired and and WiFi two separated networks, which is why I want you to connect to you switcher with software control over WiFi, to see if that could be the issue.

gliepins, 2020-10-21:
HI, here are some pics. Ask anything, hoping for your help!
[image]
[image]
[image]
[image]

AronHetLam, 2020-10-21:
That all looks good. All i can think of is that it might be an issue with the latest version. 
Try with v2.0, and let me know if that changes anything:
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/releases/tag/v2.0

gliepins, 2020-10-21:
Wait! I started to notice strange behaviour with the switcher and the network. It does not obtain ip from the dhcp server. It could be the switcher causing problems. I will get to the bottom of it. Stand by :)

AronHetLam, 2020-10-21:
Static IP addresses will probably help

gliepins, 2020-10-21:
WORKING!! 
Thank you very much! 
In situations like these it is important to rely on solid knowledge of things before messing with matters I 'm not sure of!

AronHetLam, 2020-10-21:
What was the problem, and how did you solve it?

gliepins, 2020-10-21:
Yes, I solved it! Problem was in the Atem switcher having some bogous ip settings. At first it was static ip and I didn't change it. It seemed ok because subnet was correct and it was pinging alright. But when switched Atem over to dhcp, it updated/flushed its ip settings and started to work!

gliepins, 2020-10-21:
What exactly this setting does? 
[image]

AronHetLam, 2020-10-21:
Ok, cool.

The setting is described here:
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266#modes-of-operation

You can try them out and see which one fits your use case.

closed 2020-10-21 by AronHetLam

gliepins, 2020-10-21:
Got it!
Another Q! When in preview, esp LED (not tally LED) glows blue all the time. When in program, it does not glow.
Anything worth knowing here? 
[image]
[image]

AronHetLam, 2020-10-21:
It's connected to the red pin of the second LED as I remember, but inverted signal. I find it useful for showing it has power, if eg. the LED is off.

gliepins, 2020-10-21:
I see! My concern only was if it indicates some unwanted mode or traffic which could impact power consumption, that's all.
Thanks!