AronHetLam/ATEM_tally_light_with_ESP8266 #8: No Initial Setup

maf-o, 2020-11-20:
Hello, where can i enter my Wifi ingredients?
Tried diffrent Moduls, only one i managed to get in AccessPoint, but it had SSID long long time ago. So i erased Flash + Sketch&Wifi an uploaded diffrent times no one brings me up AP Mode.
Even i am unable to tell where the magic happens in the sketch :(

I added:
`#include <TallyServer.h>
**const char* ssid     = "YourWifiName";         // The SSID (name) of the Wi-Fi network you want to connect to
const char* password = "Password";`**

then i commended out:

`   // **WiFi.config(settings.tallyIP, settings.tallyGateway, settings.tallySubnetMask);**
`
and added:
`WiFi.begin(**ssid, password**); `

Then i got IP looked up in Monitor and get access to webside so edit for my settings

AronHetLam, 2020-11-21:
The magic happens in a a function called handleSave(), which saves the WiFi settings in eeprom for you, from what you put in the website.
In setup it loads what's in the eeprom, and uses that to connect with.

It sounds like old data in eeprom might be causing you issues. Try to select erase all and reupload the sketch. That should clear everything.

You can hardcore it like that just fine, but setting the IP and WiFi settings on the website won't have any effect.

AronHetLam, 2020-11-21:
What module do you have? Maybe you selected the wrong one in the Arduino IDE, which makes it erase incorrectly. 
If D1 Mini make sure it's "LOLIN(WEMOS) D1 R2 & mini"

maf-o, 2020-11-21:
I use nodeMCU and the old one with data in was unaccessable after erasing
The other fresh out of bag doesnt work at all.
It seems to me everything is set to 255 = 0xFF

AronHetLam, 2020-11-21:
Yes, a new/erased device will have everything set to 255. It's not an issue, just make sure not to use static IP with those values (as mentioned in the guide).

The AP magic happens in the first part of the loop, and only if it doesn't connect to a WiFi after first try. If it connects to a WiFi you are able to connect to the setup website via its IP, given it has a valid and reachable IP address.
A device with erased WiFi settings should always create an AP.

maf-o, 2020-11-21:
But with everything on FF it cant connect to any network here, and accespoint is not opend.
It was only accessable with the one with old data which cant be connected and no FF

AronHetLam, 2020-11-21:
I've only worked with D1 minis, and on those it all works fine. And as it's still an esp8266 it sounds strange to me.

If it doesn't connect it should make the access point in this part of the code:

Did you accidentally comment out the part where it sets firsRun to true before setup()?

AronHetLam, 2020-11-21:
What do you see in the serial monitor?

AronHetLam, 2020-11-21:
But if you can't get it to work you can hardcore the settings in setup instead of letting it load them from eeprom. That's all done in setup().

maf-o, 2020-11-21:
With changes in my comment it works ^^

In Serial i see no SSID and everything else set to 255 (0xFF)

I made no changes  for first test.
Will have a look next week when i get new one

AronHetLam, 2020-11-21:
Great 🙂

What you see in serial is what you should see, and that should cause it to make the access point available. 

As I mentioned I only tried with D1 mini, but why it's different I don't know.

Let me know if you find anything. 🙂

AronHetLam, 2020-11-26:
Andrej seems to use a nodeMcu in the issue he posted:
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/issues/9 (his issue is irrelevant to yours).

His issue was not with the nodeMCU itself, so my guess is that you might have selected the incorrect board in the board manager when you uploaded the program. Make sure you have the correct one for your specific board. (I haven't checked which option it would be).

AronHetLam, 2020-12-21:
Any progress?
Else I'll probably close this issue soon.
🙂

closed 2021-01-02 by AronHetLam