AronHetLam/ATEM_tally_light_with_ESP8266 #9: It will not connect to my switch

Andrej65416das, 2020-11-21:
### Atem tally Light Problem

Atem I have connected to the wifi router, I have it connected to the PC.

After reading the code, it connects to wifi but does not want to connect to the switcher, even if the ip address matches the atem setup.

`Connecting to switcher...`
`Switcher IP:         192.168.10.240 `

`Sending connect packet to ATEM switcher on IP 192.168.10.240 from port 51822  `
`Connection to ATEM Switcher has timed out - reconnecting!  `

Any advice on where the problem might be?

AronHetLam, 2020-11-21:
Thanks for posting it here. 🙂

At the moment I can think of two possible issues.
 1. You have too many devices connected to the switcher already, making it refuse the tally to connect ATEM switchers only allow for 5-8 connections, depending on the model.

Or,
2. Your router separates WiFi from ethernet. Some routers do this by default for security reasons, to make guests unable to connect to wired devices.
Are you able to connect to the switcher with software control over the same WiFi and IP settings as you try to connect the tally through? If you can't, you need to make sure WiFi and ethernet are on the same Lan and subnet.

AronHetLam, 2020-11-21:
Also, you can post the pictures here as well. 🙂

Andrej65416das, 2020-11-22:
### I am attaching pictures
------------------------------------------------------------------------------------------------------------------------------------
[image]
------------------------------------------------------------------------------------------------------------------------------------
[image]
------------------------------------------------------------------------------------------------------------------------------------
[image]

AronHetLam, 2020-11-22:
I see your PC is conencted to both WiFi and Ethernet, and it looks like your Atem is connected through USB.

What happens if you try to connect software control to Atem over WiFi only? (Disconnecing your pc from ethernet and usb).

Andrej65416das, 2020-11-22:
After disconnecting the Ethernet and usb cable, my Atem failed to connect via WIFI

AronHetLam, 2020-11-22:
That's what i suspected, which means your network isn't set up correctly. As mentioned, your WiFi and Ethernet is separated by your router. In the routers settings you need to make sure WiFi and Ethernet share the same Lan, which makes devices on WiFi able to connect to devices on ethernet. This is done differently on different routers, so you'd need to consult your routers manual if you don't know where to make the changes.

Andrej65416das, 2020-11-22:
Any tutorial or video on how to do this?

AronHetLam, 2020-11-22:
What router do you have?

Andrej65416das, 2020-11-22:
###  TP LINK Archer C2

AronHetLam, 2020-11-22:
Look at page 29-31 in the manual, under wifi advanced and guest network.
https://static.tp-link.com/1910012133_Archer%20C2_V5_UG.pdf

there are a few settings about isolating WiFi clients.

Andrej65416das, 2020-11-22:
I tried to look at the switcher manual, but I didn't find anything there and I still didn't fix it. Any other ideas or suggestions on how to fix it?

AronHetLam, 2020-11-22:
Do you mean router manual?

Is your WiFi a guest network?

AronHetLam, 2020-11-22:
[image]

Andrej65416das, 2020-11-22:
[image]

AronHetLam, 2020-11-22:
And the 2.4 GHz network is the one you're using?

Then I must say I'm not sure why the router doesn't route the traffic from WiFi to ethernet. Maybe some IP settings on the router.

To me it seems a bit strange that your ip's all are 192.168.10.xxx and not 192.168.1.xxx, but I assume that's something you have changed yourself.

Maybe post some screenshots of LAN and DHCP settings

Andrej65416das, 2020-11-22:
Yes, I use a 2.4GHz connection, is that a problem? I changed the ip addresses myself, can this also be a mistake?

[image]
[image]

AronHetLam, 2020-11-22:
2.4 GHz is what the tallys use, so that's fine. 🙂

IP settings looks fine to me, but something is making it block the traffic from WiFi to ethernet. I don't have a tplink myself (the pic before was from their emulator), so I'm not sure what does it. 
What is igmp snooping? I guess that isn't it, but I'm out of ideas, other than updating and # resetting it.

Else you could make a post on tplinks site.

As mentioned, as soon as you can connect to the Atem with software control over WiFi, the tally should be able to work as well.

AronHetLam, 2020-11-22:
One thing I dont know if you tried - are you able to connect to the Atem with software control with ethernet cable only?
If not, we've been looking at it from the wrong angle, and it's an issue with the Atem's network setup.

Andrej65416das, 2020-11-22:
Atem I can also connect via eternet but I have to be connected in the router, as soon as I go directly from atem to the PC via eternet, the connection will be interrupted after a while.

AronHetLam, 2020-11-22:
If it works through the router, the Atem connects to it correctly. In that case I'm confident it's a n issue with the router blocking WiFi traffic from ethernet.

As mentioned the last suggestions I can come up with now is update, reset and or restart the router.
Or maybe try with another router if you have one.

I'm sorry I can't help you out more than that. But it seems to be an issue other than with my project/code.
I hope you get it working.

Andrej65416das, 2020-11-22:
What router do you use?  Or what would you recommend on which you know that tally lights Atem works

Andrej65416das, 2020-11-23:
I want to ask will your code work for me even if it has a NodeMCU ESP8266 board. Now I'm looking for where the problem might be everywhere.

AronHetLam, 2020-11-23:
First step for you is to get software control to connect to the Atem over WiFi, as the tallys connects the same way as software control does.

The Node MCU version should work - the pictures you sent me of the ESP settings and serial print looks fine. 

I use Ubiquity gear, but the TP link should definitely be able to do it... I just can't see why it's not. As i said, you can try to make a post on their site about why you cant get communication between devices from WiFi to Ethernet.

Andrej65416das, 2020-11-24:
**Hi,** I replaced the router and it works for me,
**Thank you for your willingness to solve the problem**

AronHetLam, 2020-11-24:
You're welcome 🙂
Good to know you got it working.

closed 2020-11-24 by AronHetLam