AronHetLam/ATEM_tally_light_with_ESP8266 #10: Neopixel support

AlexCruz02, 2020-12-10:
im requesting this because most of the people uses ws2818b leds or ws2811

**(and i only have ws2812b leds so Thanks)**

Alex Cruz

AronHetLam, 2020-12-15:
I'm not entirely sure with what you mean by:
> im requesting this because most of the people uses ws2818b leds or ws2811
As there currently is no support for addressable LED's at all, and ws2818b and ws2811 seems to be addressable?
Only Analog RGB common cathode LED's are supported.

It should be simple enough to implement the FastLed library (Neopixel support) into the project, but It's not something I've worked with before. And i don't have addressable LED's myself.

I don't know how much Arduino you know? If you've worked with Neopixels before, setting the LED colors happens in line 260 to 314 in the Arduino file (v2.1). Feel free to fork the project and see if you can implement it.

It will be a while before I have time to look at it, if it's something I ever want to add, as I wish to keep the project as simple as possible. 😊

bfbones, 2020-12-19:
I've just used this project as a basis for a fully fletched tally light system including cases, 18650 batteries for power with fully working battery monitor and WS8212B LEDs. I will provide some PRs and more in the next days.

AronHetLam, 2021-01-02:
> 
> 
> I will provide some PRs and more in the next days.

Wondering if you forgot this? Just a reminder 😊

bfbones, 2021-01-04:
Sorry, there was alot going on last days, like for everyone.. ;)
I've just forked your project and pushed my stuff. I've also updated readme and wiki:
https://github.com/bfbones/ATEM_tally_light_with_ESP8266

Unfortunately git completely failed doing that diff (or I failed?!), so that I think we are not able to cherry pic some stuff from that code for some improvements in that project over here. Maybe it's best to keep both projects as their are, cause they are alot different already.

AlexCruz02, 2021-01-06:
so I'm having various problems like the ATEMmin wasn't supported and i fixedit but the **ws2812** doesn't work i mean its not turning on.

bfbones, 2021-01-06:
Can you give any detailed problem description?
Otherwise we cannot help. 
And please open a issue in my repo if it's regarding my ws2812 implementation, otherwise this really gets out of hand here.

AlexCruz02, 2021-01-06:
> Can you give any detailed problem description?
> Otherwise we cannot help.
> And please open a issue in my repo if it's regarding my ws2812 implementation, otherwise this really gets out of hand here.

i cant open an issue on your repo because its **Forked**

AlexCruz02, 2021-01-06:
so i attached the wemos d1 mini to an 15 led strip and it turned on the led number 12 and it would be nice to get this fixed

AlexCruz02, 2021-01-06:
> so i attached the wemos d1 mini to an 15 led strip and it turned on the led number 12 and it would be nice to get this fixed

its working well but in the 12th diode

bfbones, 2021-01-06:
Dude, I cannot understand what your problem is..
Did you changed the `NUM_LEDS` in the arduino file to 15?

AronHetLam, 2021-01-06:
> 
> 
> Sorry, there was alot going on last days, like for everyone.. ;) ...

No worries, what's there looks good @bfbones. Maybe It's possible to make a version that makes both types of LEDs work, and have some simple settings for amount of LEDs etc on the webpage. :)

bfbones, 2021-01-06:
That would be a good idea. If you want you can prepare the settings page and basic implementation, I can provide the WS2812 handling by PR then.

AronHetLam, 2021-01-06:
@AlexCruz02, how is it you want your 15 LED strip to work? Do you want all LEDs to act as a tally for a single input (as the one LED my version has), or do you want the 15 LEDs to act a 15 individual tally lights (which i think @bfbones' version does)?

From looking at @bfbones' code i think the 12th (last) LED indicates what state the D1 mini is in, and battery power when it's connected to an ATEM. the other 11 LEDs seem to me to work as tally ligths for input 1-11 (am i right @bfbones?)

Also @AlexCruz02, for a 15 LED strip you need to change `NUM_LEDS` from 12 to 15 
[image]

bfbones, 2021-01-06:
No, it's same to yours. One Tally Light->One Input->One LED strip. Of course each D1 mini is checking one camera input. It's only that LED 1-11 show all the same, same time (for being brighter and a better glowing angle). LED 12 shows status (like yours does, but with an additional "on and connected"-status) and blinks, if battery voltage is too low.
If you put NUM_LEDS to 15, your 15th LED will be the status LED by design.

bfbones, 2021-01-06:
Here are some pictures of my tally light solution (will give a better explaination of what I did there): https://imgur.com/a/N1dbiz0

AronHetLam, 2021-01-06:
I see, the pictures cleared things a bit 😊

> If you put NUM_LEDS to 15, your 15th LED will be the status LED by design.

I don't think that's the case, as you hardcoded it to `leds[11]`

I think one of the error @AlexCruz02 mentions is this: 
[image]
Did you add `getTransitionMixRate(0)` to `ATEMmin`? I double checked both our repos, it's not there, so I don't know how you get it to compile.

Also, why do you need the tally lights to change the preview bus. on the ATEM? I assume that's what this code does:
[image]
If you don't want the other lights to show preview, you should be able to select the "program only" mode.

bfbones, 2021-01-06:
> I don't think that's the case, as you hardcoded it to leds[11]
> Did you add getTransitionMixRate(0) to ATEMmin?
Oh yes, sorry, I forgot to adapt that accordingly. I will push a update that later.

> Also, why do you need the tally lights to change the preview bus. on the ATEM?
I wrote that in the readme.
The reason why I just don't simply use program only cut bus is, that I still want to use the standby/preselect cam option in our livestream productions. Every cam operator needs to know when he is going to be live. That option simply doesn't exists in program only cut bus mode.
The preview mode in my opinion brings a small flaw with it. I will show it as an example:

Normal ATEM Mini preview cut bus mode:
Cam 1 LIVE (=red), CAM 2 STANDBY (=green)   -->  Switch to CAM 2 LIVE by CUT button  -->  CAM 2 gets LIVE (=red) and CAM 1 goes STANDBY (=green).
Thats the normal operation of the ATEM.

This brings a problem: The camera operator which just was live now thinks "what, I was live, now again directly on standby?!". Due to we do very dynamic livestreams I do not have time as the livestream operator to always "disable" the cam which just got the standby cam by this design, so I implemented a handling into the Tally Light which does that:

Cam 1 LIVE (=red), CAM 2 STANDBY (=green)   -->  Switch to CAM 2 LIVE by CUT button  -->  CAM 2 gets LIVE (=red) and CAM 1 goes STANDBY (=green)  -->  CAM 2 is set as STANDBY by the tally while being LIVE (=red)  -->  STANDBY on CAM 1 is off.

AronHetLam, 2021-01-06:
Ah, I see what you mean. However, be aware that my tally server library doesn't support it. Also, it's probably not something i will put in one of my releases, as it would differ from how "professional" tally lights work. At least I've never heard of tally lights that switch your mix bus for you 😉

I pulled your version into a new branch in my repo. Theres a version that should support both regular LEDs as before, and (most of) your FastLED implementation. I hardcoded `mixRate` to 1000... 
The branch is [NeopixelSupport](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/NeopixelSupport).

AronHetLam, 2021-01-06:
Another thing I noticed is your timers. I don't see you use millis(), but still have variables including millis in their names. To me it seems like you're just counting loops, and not actual time. Or am i missing something?

bfbones, 2021-01-06:
Thats correct. I tried a few several methods using time, of course also millis and somehow it all messed up various things, like correct fastled handling, incorrect working timer for that standby lite changing and so on. Further millis got slower after implementing the fastled handling. That gave me the feeling that millis is also just some basic counting as there is no real clock, but that are just my two cents, that was my first time working with those chips.
Coming from other languages I then just created a basic and simple counting way of messuring time, which works out really good also in real world usage, never had any timing problems.
I am besides that of course into doing it right with millis, so if you see a way there I am up to a better implementation.

bfbones, 2021-01-06:
> Ah, I see what you mean. However, be aware that my tally server library doesn't support it. Also, it's probably not something i will put in one of my releases, as it would differ from how "professional" tally lights work. At least I've never heard of tally lights that switch your mix bus for you 😉

Sure, that is not how professional tally lights work, but as the ATEM mini does, is also not like a professional video mixer is working. Out of my experience you just bring a cam live and there is no other cam on standby without touching anything.
That is maybe just a problem of the ATEM mini and it's limited configuration options and I was searching for a easy way to workaround that. Maybe I remove that also out of my repo, cause thats workarounding a special case.

> I pulled your version into a new branch in my repo. Theres a version that should support both regular LEDs as before, and (most of) your FastLED implementation. I hardcoded `mixRate` to 1000...
> The [...]

Already saw that, I will provide some fixes and the removal of that special case later. Thanks alot!

AlexCruz02, 2021-01-06:
i kinda fixed the problems [here](https://github.com/AlexCruz02/ATEM_tally_light_with_ESP8266/tree/master/ATEM_tally_light) its my fork

AronHetLam, 2021-01-11:
What needs to be done before releaseing Neopixel support:

- [ ] Remove deepsleep on low battery (Shuts down boards not using voltmeter, but eg USB power)
- [ ] Remove functionality to change preview bus (or make it optional in settings).
- [ ] Refactor timers to use millis() ([needs testing - maybe FastLED messes with interrupts](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/issues/10#issuecomment-755497094))
- [ ] make status colors consistent with earlier versions where possible.
- [ ] Test FastLED performance impact, and limit amount of Neopixels accordingly.
- [ ] Update README and WIKI accordingly
    - [ ] Add circuit diagram/description for adding Neopixels.
        - [ ] _Maybe add circuit diagram for battery use and voltmeter_
    - [ ] Add new color description
    - [ ] Add status LED description.
    - [ ] Add Neopixel settings descriptions.

These were the things i could think of. Let me know if I missed something. 
Remember, my goal is to keep it as simple as possible for non-coders to solder, program and build this project. No coding should be required by my releases. 😊

AlexCruz02, 2021-01-23:
Good and bad news 

- Its connecting to the switcher 
- It's not lighting up when I change the camera and I'm using a ATEM 1 M/E production 4K

goudi76, 2021-02-09:
Hey, I really like the Idea using Neopixel. I´m not so good in writing software but I couldn´t find code where the Neopixel are running. Will it be possible to have Neopixel only? 
That would decomplex a lot using more LEDs to get it brighter.
Another Question: @bfbones have you uploaded your 3D Model of the case? I´m serching exactly for that but I won´t use battery support. I think it´s better using supply power.
Putting the case on a cold shoe would be perfekt And I would like to use a stripe Maybe I need to create my own 3d print.

AronHetLam, 2021-02-09:
@goudi76, did you look in the [NeopixelSuppoert](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/NeopixelSupport) branch? There you'll find the work in progress version of Neopixel support as it is at the moment.
See if you can get that to work. Be aware that you need to download the FastLed library for it to work. You can do so within the Arduino Library manager.

AronHetLam, 2021-02-09:
@goudi76, I just uploaded a new version to the [NeopixelSuppoert](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/NeopixelSupport) branch, containing what should be a fix to the first two points on my list above. This makes you able to use it without a battery.

goudi76, 2021-02-09:
I have tried to put it on my Wemos board and it seems to run but I cannot see the Neopixel changing. It was changing once but after I power down the Wemos and powered it up again all neo Pixel stay dark. I have used a used board and the stipe is soldered to D2 and I have changed the output to D2 in the code before I compiled it.
I guess that is not how it should be. I am also not sure if rgb is correct for my stripes are working in grb mode. I could not find to put in the code where to hard fire up the stripes to make sure that it is working proper.
I hope this helps a bit?

goudi76, 2021-02-10:
Is there a way (serial monitor command) or command on the webpage easy to implement to do such trouble or tests? 
Maybe in more steps: first step the hardware Test LEDs second the whole network.
Even if there is no Atem available it would be nice if I could set on Wemo to "program2, and/or Preview4" and a nother Wemo recognize it so Wlan Settings can be checked without a Atem beeing present?! - this is just an idea - maybe something like that is already available (but I couldn´t find it in the code)

AronHetLam, 2021-02-10:
D2 is used for the green color for the first regular LED, maybe that causes an issue.
In line [52](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/6a36af361e743c7ec202eb8abc6d2857d7b7c756/ATEM_tally_light/ATEM_tally_light.ino#L52), [73-74](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/6a36af361e743c7ec202eb8abc6d2857d7b7c756/ATEM_tally_light/ATEM_tally_light.ino#L73) and [166](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/6a36af361e743c7ec202eb8abc6d2857d7b7c756/ATEM_tally_light/ATEM_tally_light.ino#L166) the LED strip is defined as CRGB.
In line [169-173](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/6a36af361e743c7ec202eb8abc6d2857d7b7c756/ATEM_tally_light/ATEM_tally_light.ino#L169), sets the strip off and the status LED to yellow at the moment. Changing this and putting a long delay or infinite loop makes you able to test the strip.

For testing I once made this version for when I don't have a switcher around. By pressing enter in the Serial monitor, it cycles through no tally, preview and program for connected tally lights. It's based on an old version, and should be used as test server/switcher only. 
[ATEM_tally_Server_Serial_Control.zip](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/5956523/ATEM_tally_Server_Serial_Control.zip)

AronHetLam, 2021-02-10:
Also, if you have had the released version of the program on there, maybe you need to select "All Flash Contents" in Tools -> Erase Flash, to make sure settings between versions don't collide.

goudi76, 2021-02-10:
well I have also changed D7 to D2 and  D2 to D7 - so that should have been save.
Thanks for your explanations where I do find all the stipe settings.
Maybe your second hint is most important and I have had some old content stored in the ESP. I will check that later.

goudi76, 2021-02-10:
May I kindly ask you, did it already run with a stripe of Neopixel? I get easily fastled examples to running condition using the same board.
I´m a little bit missed up in your code.
But with this code I can´t get any light out of the stripe.

bfbones, 2021-02-10:
I am running 4 devices without any issue with 12 in a row.

AronHetLam, 2021-02-10:
@goudi76, I'd need a bit more info.
What do you see in the serial monitor?
Did you specify an amount of neopixels and a status LED in the settings website?
Could you paste your example code? You mention you use other LEDs than CRGB.

I removed functionality easterday, so that it wouldn't turn of when not having a battery, maybe I broke something else too... But as mentioned earlier in this thread, I don't have a led strip myself. The Neopixel support that's there is thanks to @bfbones 🙂

bfbones, 2021-02-11:
> have you uploaded your 3D Model of the case?

I habe the model files, but my case has been done by a good friend of mine and got my company branding. I will ask him, if he can provide me a branding free version which I can share.

AronHetLam, 2021-02-11:
I did some digging to see if I could find an issue, but the only real one i ran into was that it went to deepsleep on earlier versions, because I don't have a battery attached. As soon as I uploaded the latest one I committed that was solved.

With a regular LED on the data pin i can see it change the intensity when I change between off, preview and program, clearly indicating that it's sending different data streams trough the data pin.  

I just now committed a bug fix that made it crash if you had 0 Neopixels, and no status LED selected in the settings. But I don't think that's the issue you're having @goudi76 .

goudi76, 2021-02-11:
Thanks a lot for having a look inside the code..
let me copy a part of the code where I put in some changes for TS (Trouble shooting)
I used the code which is uploaded from today!
`   int ledsOffset = 0;
    if (settings.neopixelStatusLEDOption != NEOPIXEL_STATUS_NONE) {
        ledsLength = settings.neopixelsAmount - 1;
        if (settings.neopixelStatusLEDOption == NEOPIXEL_STATUS_FIRST) {
            ledsOffset = 1;
            FastLED.addLeds<WS2812, DATA_PIN>(statusLED, 1);
        } else {
            FastLED.addLeds<WS2812, DATA_PIN>(statusLED, ledsLength, 1);
        }
    } else {
        ledsLength = settings.neopixelsAmount;
    }

    leds = new CRGB[ledsLength];
    if(ledsLength > 0)
        FastLED.addLeds<WS2812, DATA_PIN>(leds, ledsOffset, ledsLength);

    setSTRIP(LED_OFF);

    //Set LED Status
    Serial.println("LED set to yellow");
    setStatusLED(LED_YELLOW);
    FastLED.show();
  delay(10000);
  Serial.println("post delay");
    Serial.println(settings.tallyName);
    //Serial.println(sizeof(settings)); //Check size of settings struct
    if (settings.staticIP) {
        WiFi.config(settings.tallyIP, settings.tallyGateway, settings.tallySubnetMask);
    }
`
in the serial Monitor following is showed:

⸮⸮  ⸮   ⸮⸮ ⸮ ⸮⸮⸮ ⸮   ⸮ ⸮⸮ ⸮ ⸮⸮  ;d lܟ| ⸮l⸮<⸮l⸮c|⸮⸮⸮⸮{⸮#⸮#⸮⸮gn⸮$go⸮⸮⸮cp⸮⸮$;l;dp⸮'⸮⸮d⸮⸮cg⸮|⸮d⸮⸮b⸮⸮'o⸮ l⸮⸮$`⸮gn$ os⸮⸮⸮g#l {⸮⸮'bl`⸮#⸮⸮l;⸮`⸮⸮o⸮########################
Serial started
LED set to yellow
post delay
⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮
------------------------
Connecting to WiFi...
Network name (SSID): 

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
########################
Serial started
LED set to yellow
post delay
ESP-09E695
------------------------
Connecting to WiFi...
Network name (SSID): Markus xxxx
------------------------
Connected to WiFi:   Markus xxxx
IP:                  255.255.255.255
Subnet Mask:         255.255.255.255
Gateway IP:          255.255.255.255
------------------------
Connecting to switcher...
Switcher IP:         255.255.255.255
> 
to explain the situation:
I upload the code and the first LED comes up blue.
Then I connect to the Webpage and put in my credentials.
I put the Atem LED to 2
I put amounds of LED to 6
Status LED is first
save
the Result is that everything is black on the Stripe.
I hope this helps more?

goudi76, 2021-02-11:
I don't know how it works to have the correct format - sorry for the mismatch upstairs.

goudi76, 2021-02-11:
is there an easy way to set it to "program" or "preview" with the connection to the PC on the board?
Or do I need a second Demos with the testing Software on it?
But in any case the yellow shout flash up.
PS: I have soldered the pins around and now I do stick with the Data pin in D7

AronHetLam, 2021-02-11:
What happens if you add __GRB__ to the addLeds functions, and change setStrip to __LED_BLUE__, as shown below? You mentioned that your LEDs are GRB, and not RGB. Changing the order might make the first LED yellow after upload.
``` C
if(ledsLength > 0)
    FastLED.addLeds<WS2812, DATA_PIN, GRB>(leds, ledsOffset, ledsLength);

setSTRIP(LED_BLUE);
```
The status LED is meant to be either the last or first led of the strip, indicating the tallys state, such as connection and battery status.
The setStrip sets the rest of the LEDs, which should only work as tally.

If this doesn't change your results at all, please provide one of the working examples you've used.

Also, I notice your IP settings are bad. If you do not set valid static IP settings you should uncheck the static IP checkbox, så that it can get an IP from DHCP.

> is there an easy way to set it to "program" or "preview" with the connection to the PC on the board?
> Or do I need a second Demos with the testing Software on it?

At the moment you'd need a separate Wemos to act as a switcher with the test software.

bfbones, 2021-02-12:
> What happens if you add **GRB** to the addLeds functions, and change setStrip to **LED_BLUE**, as shown below?

My RGB strip was actually a GRB, but it was working with swapped colors, so I don't think this is @goudi76 's problem.

goudi76, 2021-02-12:
Hy Thanks @bfbones. I yust want to explain that by myself but you have been faster....
I have checked between GRB and RGB setting using the example from fastled. Changing "G" and "R" swaps the color green/red.
Ich you show "green" with the wrong GRB / RGB setting it shows "red" instead of green. But blue stays blue in any case. So Yellow has both, red and green - the yellow may look not nice but not dark. I guess there is still another problem. I want to try later on to set it to "program" with another Wemos and your Test Library you have send out. See what hapenes. Is it reported in the Serial Monitor if it changes the program setting? 
I do still not have a RGB Led (I have ordered them) to put them to the other DOs for testing.

AronHetLam, 2021-02-12:
@goudi76 My suggestion was more to see if the strip would turn on, and if the first LED would have a different color than the rest of the strip. I have a feeling that the statusLED might be overwritten when the rest of the strip is added, which would explain why at first one led is lighting up, when the amount of Neopixels is 0. If it is overwritten setSTRIP(LED_OFF) would turn off all LEDs, explaining why you aren't seeing anything. You never get to the point in the program where the strip is turned on (except for the status LED), while you don't have it connect to a switcher (or test software). setSTRIP(LED_BLUE) would let you be able to see if it actually does set the strip color, and which LEDs it sets. 

What you should see at your 10 sec delay, with 6 Neopixels and First LED as status LED, is that the fist LED is yellow, and the last 5 LEDs are blue. What I suspect you'd see is that the first five might be blue, and the last one is off. In that case we need to make a change to how we load the strip.

@bfbones did you ever try a version with Neopixel settings on the website? If so, what was your results?

bfbones, 2021-02-12:
Which Neopixel Settings do you exactly mean?

AronHetLam, 2021-02-12:
I added these after pulling your version into my NeopixelSupport branch:
[image]

goudi76, 2021-02-12:
@AronHetLam maybe we can arrange a meeting with teams? That would give us the posibility to share cam and screen. Please send me a pn. Then we hopefully will find a time slot?

goudi76, 2021-02-12:
I have set up the server and tried to toggle the status.
What shell I say?
I have put in all correct network setup data. How can I keep them and overwrite the .ino file?
When I push (CR) setting used in the serial monitor of the server, the LED Strip toggles black, gren,red, black I guess it must be the other way arround? So I would have to change the GRB settings.
But: The Yellow start did not light up and the Status LED did not work anymore. 
[image]

AronHetLam, 2021-02-12:
If you select "Only Sketch" in Tools -> erase flash, it'll keep the settings. across uploads.

I looked more thoroughly at how fastLed works, and I found out that I'm indeed adding the LEDs in a way that the strip overwrites the status LED. I'm working on a fix, and will upload it as soon as it's done. 🙂

goudi76, 2021-02-12:
I do not understand this line:
[image]
as far as I have the right understanding it says use ws2812 stripe  ,  DataPin > (LED...

in the fastled example this Line is a bit different:
it says use ws2812 stripe  ,  DataPin, GRB> (LED...

Is there something missing? Or where should I turn RGB to GRB?

AronHetLam, 2021-02-12:
You can add it optionally. If not, I assume it defaults to RGB. just remember the comma.

goudi76, 2021-02-12:
`FastLED.addLeds<WS2812, DATA_PIN, GRB>(statusLED, 1);`
Thanks. This turns the color to be right. Maybe these lines needs to be set up like I have done it.

AronHetLam, 2021-02-12:
Just uploaded the fix. That should make the settings and strip initialization work as intended.

> `FastLED.addLeds<WS2812, DATA_PIN, GRB>(statusLED, 1);`
> Thanks. This turns the color to be right. Maybe these lines needs to be set up like I have done it.

Hardcoding it like that would make it work correctly in your case, but not for others, depending on the type of LED strip they have. 
The best solution for non-coders is to add it to add it as settings in the website, but I'd have to think of the best way to do so. However, I assume most people knowing what a Neopixel is would be able to get it set up as it is now. 😊

goudi76, 2021-02-12:
I have tested the new code thanks for that.
I have some Ideas: 
it would be good to add that a Stripe can be choosen and as well the way of color (RGB, GRB)
As well it would be good to add a dimming using the command "brightness" for dark  surrending on the website.
At the first start a blue LED accures at LED nr. 5
I set the amounts of LEDs to 6.
Using the test Wemos only 5 LEDs are lighting up (Step2, green) (LED 1-5) Status LED is last.
When I put the Status LED as the first LED, only 4 LEDs are lighting up (Step2, green)

AronHetLam, 2021-02-12:
I can't see why the whole strip wouldn't work...

uncommenting line [440-441](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/cb96b5bb25af23364b47250fb3fda513f7121c2c/ATEM_tally_light/ATEM_tally_light.ino#L440), [454-455](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/cb96b5bb25af23364b47250fb3fda513f7121c2c/ATEM_tally_light/ATEM_tally_light.ino#L454) and [458-469](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/cb96b5bb25af23364b47250fb3fda513f7121c2c/ATEM_tally_light/ATEM_tally_light.ino#L458) makes it print RGB values for the whole strip, every time it is changed. 
This is what it looks like for me with 6 pixels and status led first:

```
Serial started
Tally:
0, RGB: 0, 0, 0
1, RGB: 0, 0, 0
2, RGB: 0, 0, 0
3, RGB: 0, 0, 0
4, RGB: 0, 0, 0
5, RGB: 0, 0, 0

Status:
0, RGB: 255, 255, 0
1, RGB: 0, 0, 0
2, RGB: 0, 0, 0
3, RGB: 0, 0, 0
4, RGB: 0, 0, 0
5, RGB: 0, 0, 0

ESP-166D94
------------------------
Connecting to WiFi...
Network name (SSID): Het lam 2
Status:
0, RGB: 255, 255, 0
1, RGB: 0, 0, 0
2, RGB: 0, 0, 0
3, RGB: 0, 0, 0
4, RGB: 0, 0, 0
5, RGB: 0, 0, 0

Tally:
0, RGB: 255, 255, 0
1, RGB: 0, 0, 0
2, RGB: 0, 0, 0
3, RGB: 0, 0, 0
4, RGB: 0, 0, 0
5, RGB: 0, 0, 0

------------------------
Connected to WiFi:   Het lam 2
IP:                  192.168.1.50
Subnet Mask:         255.255.255.0
Gateway IP:          192.168.1.1
Status:
0, RGB: 255, 165, 0
1, RGB: 0, 0, 0
2, RGB: 0, 0, 0
3, RGB: 0, 0, 0
4, RGB: 0, 0, 0
5, RGB: 0, 0, 0

Tally:
0, RGB: 255, 165, 0
1, RGB: 0, 0, 0
2, RGB: 0, 0, 0
3, RGB: 0, 0, 0
4, RGB: 0, 0, 0
5, RGB: 0, 0, 0

------------------------
Connecting to switcher...
Switcher IP:         192.168.1.37
```

goudi76, 2021-02-14:
Thanks alot for your work. Now I have found everything working well.
The Blue and Yellow Status LED is as first and stays there correctly.
I guess it is a little bit "nervous" and "felt" not showing the correct status. It was yellow but recognizing a new mixer status.
What was the failure? I have probably downloaded the wrong code because I couldn´t find the lines to coment out for the Serial output. After I found the correct code everything was fine.
Thanks alot!

AronHetLam, 2021-02-14:
Good to hear. At the moment it still thinks there's a battery (As that's what @bfbones' version has), which makes the status light blink to indicate low power. You can ignore that part. 😉

AronHetLam, 2021-02-17:
Check out release [v3.0](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/releases/tag/v3.0). I hope I got everything. That should make for a useable version, supporting Neopixels.

closed 2021-02-17 by AronHetLam

AronHetLam, 2021-03-23:
> 
> 
> Here are some pictures of my tally light solution (will give a better explaination of what I did there): https://imgur.com/a/N1dbiz0

@bfbones, If you don't mind sharing your 3D files for your case in your fork, I could link to it from the wiki for others to find. 

https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/wiki/DIY-cases