AronHetLam/ATEM_tally_light_with_ESP8266 #11: other systems support

gliepins, 2020-12-19:
Hi!
Can you give some directions on how to implemet other systems support besides BlackMagic?
First, Vmix! Can you tell what parts of the code should be altered to add support for vmix? And if that works, how to further add support for more systems?

AronHetLam, 2020-12-19:
It's a big request for multiple reasons. The main one being that this solutiom heavily depends on Skaarhoj's Arduino libraries for ATEM, which take care of all the communication with the ATEM. Also, adding Vmix support could be better suited to do in a separate project, as you would waste a lot of unnecessary resources on the small WiFi modules by supporting both at once, when only one connection would ever be used.

I don't know how much Arduino you know, but I'd suggest you'd take a look at Vmix's own Arduino tally light solution. You'd probably be able to somehow get it working on an ESP (without promising anything).

Tally arbiter is another open source project for tally lights, which i know has a solution for ESP's as well.

Companion is another good source for inspiration. It's for use with streamdecks, but they have a ton of libraries for all kinds of video systems.
However, both of those run on NodeJS, so you'd need something like a raspberry pi as server.

This project won't add support for other systems anytime soon, as I don't have other systems myself, and because there are other open source tally systems out there covering a lot of cases.

AronHetLam, 2020-12-19:
But to actually answer your request for direction, connecting to the ATEM and getting the tally data from the ATEM library is all done in the setup and loop functions.

Once the connection to the ATEM is established the LED colours are set in the loop as well, using the setLED1 and setLED2 functions. The code is commented, so you should be able to see what's going on where 🙂

gliepins, 2020-12-20:
Thank you for your answer. Clears a few things! 
Case closed!

closed 2020-12-21 by AronHetLam