AronHetLam/ATEM_tally_light_with_ESP8266 #12: card not working

AlberONE, 2021-01-03:
hi, I followed the instructions on mounting the card and after regularly programming the card I tried to see if everything worked. After about an hour of switching on the red LED, the card broke. I thought I had badly powered the card with the 5 volts pin, so I redid the whole sequence on a new one by feeding it with the micro USB. after an hour the new one broke. Do you have any suggestions?

AronHetLam, 2021-01-03:
Sounds odd. Powering the D1 mini with 5v through the 5v pin is the same as powering it through USB, so unless you accidentally powered it with too much voltage both ways, that's not the issue.

What LED do you use, and how many amps does it draw? Mine draw around 20 mA. The esp8266's pins can't handle much more than that. I even think the datasheet rates them at 12 mA only, however mine are still fine after half a year of weekly usage.

I assume you have used the correct resisors, and that "broke" means that you aren't able to reprogram it or see anything in the serial monitor, and that it isn't just the LED that doesn't work.

AronHetLam, 2021-01-03:
Maybe you could add a picture of the chip.

AlberONE, 2021-01-04:
Salve,
Il problema si verifica dopo circa un’ora dall’accensione della luce rossa. Volevo verificare quanto le batterie rendessero autonomo il kit. Quindi ho attivato la luce rossa e dopo un’ora la luce è diventata rosa. Ho provato a riprogrammare la scheda ma non ci sono più riuscito.
Allego il fotografico
Grazie
Alberto

AlberONE, 2021-01-04:
Hi, 
the issue happens one hour after the red light turns on. I wanted to prove the duration of the battery, so I activated the red light but it turned pink after an hour or so. I tried to program the card again but I didn't managed to. I am attaching a photo. Thank you in advance. Alberto

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]
[image]

[image]

AlberONE, 2021-01-04:
this is resistors

[image]

AronHetLam, 2021-01-04:
I assume you closed the issue by accident, so I reopened it 🙂

To me the first message sounded like you fried the chips, but that's not the case. The pink light indicates that it looses the connection to the ATEM. On the settings website the status should reflect that state.

From looking at your pictures it looks loæike you wired it correctly.

Why it would loose it's connection to the ATEM after an hour I don't know. I'd have too look into that once I have time.
Does it reconnect if you turn it off and on?
Did you select the correct COM port and board in board manager when you tried to (re)program it? (COM port can change when you plug it in again)

AronHetLam, 2021-01-04:
This is what I have selected.
[image]

jmvb-cl, 2021-01-12:
OFF topic quick question!

Can I use the "PRO2 version, with external antenna?

https://www.amazon.com/-/es/D1-Mini-ESP8266-conector-Internet-Desarrollo/dp/B072Q4LHDL

Thank you!

AronHetLam, 2021-01-12:
@jmvb-cl it should work, just select the correct board before you upload it.

jmvb-cl, 2021-01-12:
The relay use the ESP01 board, so I guess I should use "Generic ESP8266 module"

This is the pinout

[image]

closed 2021-02-26 by AronHetLam