AronHetLam/ATEM_tally_light_with_ESP8266 #15: Add "On Air" mode

AronHetLam, 2021-01-11:
Functionality is already added to the repo's ATEMmin lib, and just needs to be implemented in ATEM_tally_light.ino and the settings webpage.

- [ ] Add "On Air" option to settings.html
- [ ] Add "On Air" mode to ATEM_tally_light.ino
- [ ] Add documentation to README and WIKI

jmvb-cl, 2021-01-14:
What if I don't use the second RGB LED, and instead I connect a red LED in D4?
Would be the same functionality?

AronHetLam, 2021-01-14:
If you at the moment just put a red LED on D4 you'd only see tally live status for the selected input. As the initial post states, "On Air" mode (whether the ATEM is streaming or not) has not been added to the Arduino code yet.

jmvb-cl, 2021-01-14:
So, the "on air mode" is about the whole tally lights will have a red light when the streaming is happening?
That is for "atem mini" users, right?

I thought that you were referring to the individual on air lights, thats who the presenter on people in the stage would see, to identify which camera look

[image]

AronHetLam, 2021-01-14:
The tally lights per individual input (camera) would switch between program and preview - telling the operator whether his cam is output, or about to be output. That's what this project does at the moment, on both LEDs.

"On Air" refers to whether the ATEM is streaming or recording content or not. So yes, as far as i know only the ATEM mini pros supports this. This is the feature this issue is about implementing, so that you use this project as an "On Ari" sign if you wanted too.

closed 2021-03-14 by AronHetLam