AronHetLam/ATEM_tally_light_with_ESP8266 #16: Use of D1 mini pro and esp8266 relay module

AronHetLam, 2021-01-12:
OFF topic quick question! (2 actually)

1.- Can I use the "PRO2 version, with external antenna?

https://www.amazon.com/-/es/D1-Mini-ESP8266-conector-Internet-Desarrollo/dp/B072Q4LHDL

2.- can I use a Wifi Esp8266 relay module?

[image]

Thank you!

_Originally posted by @jmvb-cl in https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/issues/12#issuecomment-758609768_

AronHetLam, 2021-01-12:
The relay use the ESP01 board, so I guess I should use "Generic ESP8266 module"

This is the pinout

[image]

_Originally posted by @jmvb-cl in https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/issues/12#issuecomment-758644194_

AronHetLam, 2021-01-12:
@jmvb-cl The D1 mini pro should be easy to use, as it's just a matter of selecting the correct board in Arduino, and maybe change a few pin definitions.

Yes, selecting "Generic ESP8266 module" should be the right one. make sure to manually select the right flash size etc.
However, using the ESP01 with the relay board will require a few more changes, as the board only seems to let you use 2-4 pins for output (if you disable serial print and flash), and only has two relays. The current [release](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/releases/tag/v2.1) supports 2 LEDs, so 6 pins total. You'd need to remove/change the parts of the code that uses the last 6 pins, and only be able to work with red and green - limiting status indication.

The parts you'd need to change are:
- line 28-36; defines output pins. For GPIO1 you'd replace "Dx" with "1"
- line 86-93; defines pin modes. You'd remove the unnecessary ones
- line 260-314; sets the LEDs. You'd change this to set your two relays instead, and remove the rest, without deleting the functions.
- Probably you'd need to remove all the "Serial" statements.

AronHetLam, 2021-01-12:
Btw, the relay might be controlled by an I2C chip, which would make for some changes, that you would need to look into.

jmvb-cl, 2021-01-13:
Aron, I managed con control an Arduino type relay directly with the D1 mini, so I guess that this topic is no longer necessary (for me al least)... 

BTW the ESP01 relay is an interesting alternative for who, like me, have field monitors with build in tally in the cameras

PDT: Today I used 2 modules (only D1 mini and led, no relay) and have some wireless issues, the D1 constantly lost the connection. Sometimes, the LED turn pink, and after a while, the D1 connect and work again.

Back in home, I changed the wireless router channel, and tested it with no issues.

Next job, I'll report back

Thank you again!!

AronHetLam, 2021-01-13:
I'm not sure how much Arduino you know, but be aware you'd need an USB programmer to program the ESP01, as it doesn't have a built in USB interface like the D1 mini does.

Connection loss can very well be related to the network being busy, as ATEM uses the UDP protocol, which is allowed to loose data being transmitted. That coupled with a not too powerful chip can couse those issues. When the LED turns pink it's disconnecting from the ATEM, and not the network itself, pointing towards the possibility of lost data. Once the connection with the ATEM is lost it takes a little while before the ATEM frees up that connection spot. Depending on how many devices and instances of software control you have, that might cause a delay in the reconnection process.

If you'd like to try with a more powerful chip, you could try the ESP32 - it comes in many shapes and forms as the ESP8266. I don't have an ESP32 myself, but I am able to verify (compile) the code, targetting it in the Arduino IDE, with some changes. (no promises though, as I can't test it more than that at the moment)

jmvb-cl, 2021-01-13:
Aron: I have an Arduino uno to use as programer

When i had the issue, sometimes the modules LED turn white...

ESP32 like this one?

[image]

AronHetLam, 2021-01-13:
If it turns white it's probably out of WiFi range, which probably also makes it more likely to drop UDP packets when only having a weak signal. In that case I don't think an ESP32 would really help.

But yes, with a few tweaks in the code an ESP32 like that should be able to run it. 
My initial idea with making it work for the ESP32 was so that poeple could get a [m5stick ATOM](https://m5stack.com/collections/m5-atom/products/atom-matrix-esp32-development-kit)

jmvb-cl, 2021-01-13:
The odd thing about the wifi range is, one camera was about 3 meters of router and the other one 1o maybe... both so close

AronHetLam, 2021-01-13:
That does sound odd, and unfortunately I don't have a good answer for that... sorry. One of the ESPs I got just won't connect to WiFi 90% of the time - probably a bad unit. My code doesn't touch the WiFi connectivity, other than telling it to connect. With the rest of my units WiFi is only an issue when I'm out of range.

Maybe you could try out another router, but why it would be the issue I don't know.

jmvb-cl, 2021-01-14:
I changed the router channel and did some house tests, flawlessly

AronHetLam, 2021-01-14:
Well, then you know it's an issue at your studio, and not your house, ruling out the tally unit being the cause. Probably WiFi/Network at your studio environment that causes this. Not something I'm that much of an expert at.

AronHetLam, 2021-03-11:
I'm closing this, as it's been inactive for a while now.

Feel free to add a new comment if closed it wrongly.

closed 2021-03-11 by AronHetLam