AronHetLam/ATEM_tally_light_with_ESP8266 #17: What Access Point are you using? Because i cannot reach my ATEM

Hitower78, 2021-01-15:
Hi There

I wonder why "Disconnected - No response from switcher" is in the Status

i am not allone with that "wireless problem to ATEM or Blackmagic Products"
https://forum.blackmagicdesign.com/viewtopic.php?f=4&t=28383

So i am asking those you use this ESP Tally: what Access Point is ok?

i use the TP-Link WR841N. Everything works (Cable), but no wireless ATEM Controll (doublecheck IPs, Subnet, Gateway, Ping, Software...)

Thanks for Help :-)

AronHetLam, 2021-01-15:
I have ubiquity network gear from where I use it, but you should be able to get it to work with any router, really.. However, take a look at this issue #9 - Also someone trying to use a TP-link without success. We sadly didn't find out why.

AronHetLam, 2021-01-15:
Just skimmed through issue #9 again. If you've tried the steps in there and have the same results, I don't know how to solve it. In your case I would contact TP-link support, and describe the issue. I don't see what should block it. Especially not if you are able to ping the ATEM over WiFi from your PC.

I have read about some cases where it helped changing the IP settings over USB on the ATEM from eg static to DHCP and vice versa. Sometimes it seems they are broken. In my setup I've given the ATEM a static IP/lease in my router config, outside the DHCP range. At the moment I don't remember if I set the ATEM's IP settings are static or not. Probably static, but I'd have to double check.

jmvb-cl, 2021-01-19:
I have a TL-WR740N in my system

Tomorrow I'll take some screen captures of my configuration...

BTW, I don't know if I made some changes before, but my ATEM IP is 192.168.10.239...

Can you control your ATEM with your PC connected by wifi?

Hitower78, 2021-01-19:
Hi there
I check now my 2nd Accespoint: its a TP-Link TL-WA901. 
The Same: 
when i use any wireless Device (Notebook, Iphone or the Tally) and try to connect to the ATEM it doesnt work. 
No Ping, no Connection with ATEM Software
Other Stuff (companion to vmix tally) is running on both Access Points

I will try another one

AronHetLam, 2021-01-19:
Yes, i can access my ATEM via software control over WiFi. If you can't you have an issue with you network... #9 was the same problem, where at the end he switched to another router/AP, which solved it for him. I really don't know why some routers blocks the ATEM traffic  from ethernet to WiFi. I recommend contacting TP-link, with a description of what steps you have taken, and that you can access it over ethernet, and not WiFi.

Are you able to ping the ATEM over both WiFi and ethernet? And is it different when you enable/disable client isolation and likewise settings? If you can ping it over WiFi, but not connect to it with software control, that's what I would tell TP-link, and ask them why and how to solve. 😊

jmvb-cl, 2021-01-19:
[image]
[image]
[image]
[image]
[image]
[image]

Hitower78, 2021-01-19:
Hi
Thanks for your answer.
No Ping with wireless Connection on both of that Access Point to the ATEM. 
So No wireless Control. 

I didnt understand that because the rest over years is still running over years with that APs 
(Ipads to companion, Notebooks, Iphones, Widget designer and so on) 

There is no Isolation and other Settings. i think the most settings are default (i only change ssid, dhcp settings and IP)

Perhaps the 3rd AP will work. 

Thanks for your Pictures. There are the same settings on the first AP (841), i will check the other later

Hitower78, 2021-01-19:
And finally it works!
A "Good" Access Point is the Key. The LANCON is so powerful and there are so many Settings...
i need 10min for the easy Hotspot settings :-P

Thanks all! I think this Topic is ready to close :-)

Greetings from Hamburg / Germany
Daniel

AronHetLam, 2021-01-19:
To me the settings looks good, so again, I don't know what causes the problems..
Good you found a solution.
I'll close this ☺

closed 2021-01-19 by AronHetLam