AronHetLam/ATEM_tally_light_with_ESP8266 #18: D1 Mini pro (with external antenna) won't set camera other than #1

jmvb-cl, 2021-01-20:
Just received a new D1 mini pro (in order to increase the wifi reception)

Downloaded the CP210x USB driver

Choose LOLIN (WEMOS)D1 mini pro in boards

Upload the sketch as usual, flawlessly

Set the tally as usual, flawlessly

Here the odd things begin..

No matter which camera I selected, at the end, always the #1 is set.

I re-do che set up 10 times, and no changes.

BUT, If the LOLIN (WEMOS) R1&R2 is selected, the set-up finally can be done

So, be advice, if the Mini pro is used, select LOLIN (WEMOS) R1&R2

jmvb-cl, 2021-01-30:
OFF TOPIC; Just see a closed issue about the ATEM 1 M/E Production Studio 4K
I have this switch and I used the tally since the day I discover it.
So, if the user who report had a problem, that was not the previous version.

AronHetLam, 2021-01-30:
@jmvb-cl Odd, doesn't the production 4k have 21 inputs? (I only have a mini pro, so the rest is pure speculation and guessing for me 😉)

Btw, you can still comment on a closed issue 🙂

jmvb-cl, 2021-01-30:
Nope, the ATEM 1 M/E have 10 inputs,,

[image]

The ATEM 2 M/E have 20 (input 1 is HDMI or SDI switchable)

[image]

PDT: my bad... can you move this to there?

AronHetLam, 2021-01-30:
I must have misread 21 (20 active) inputs for the 2 m/e as the 21 inputs i talked about. However, there are models with 40 inputs, so the fix is still relevant for those cases.

If @alexcruz still has a problem with his setup, I'd have to look into that. If you have the same swithcer as he has, maybe you know something @jmvb-cl?

I might be able to quote these posts in issue #9

jmvb-cl, 2021-01-30:
I just follow your instructions and all was perfect

AronHetLam, 2021-02-04:
As this isn't really an issue I can dommuch about I'll close it here.

Others can still find it. 🙂

closed 2021-02-04 by AronHetLam