AronHetLam/ATEM_tally_light_with_ESP8266 #19: Good and bad news

AronHetLam, 2021-01-23:
Good and bad news 

- Its connecting to the switcher 
- It's not lighting up when I change the camera and I'm using a ATEM 1 M/E production 4K

_Originally posted by @AlexCruz02 in https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/issues/10#issuecomment-765794194_

AronHetLam, 2021-01-23:
This is another issue than Neopixel support, so I put it here.

I might know what's going on. I looked at the ATEM production 4k specs, and saw that it seems like it has 21 inputs. I think this is a case of Skaarhoj's libraries being outdated, only supporting 20 inputs.
I'll commit a version that allows for 40 inputs later today, as it seems one of the ATEM models has that many.. 🙂

AlexCruz02, 2021-01-23:
Thanks You're the best

AronHetLam, 2021-01-23:
I made release [v2.1.1](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/releases/tag/v2.1.1) which should fix the issue, if that limit was the issue. In your case you'd only need to replace the libraries, and not the Arduino code, unless you actually need tally for input source 21. That way you can keep your modified Arduino code. 

Let me know if it works. 😊

closed 2021-01-23 by AronHetLam

AronHetLam, 2021-02-02:
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/issues/18#issuecomment-770211459