AronHetLam/ATEM_tally_light_with_ESP8266 #20: LED Colors

lschaef, 2021-02-10:
Is it possible to change the assigned colors for events?

RED | Tally program & GREEN | Tally preview isn't very good for camera operators who are Red/Green colorblind.  :)

AronHetLam, 2021-02-11:
You got a good point there 😉
It is possible, but you'd need to do it in the code.
Not knowing how much coding you know, I'll direct you to line 276 in the Arduino file, where you'll find the method setting the colors. Depending on the color it changes to, it'll set the value of the three pins accordingly with the methods digitalWrite(...). A 1 turns a pin/color on, and a 0(zero) turns it off.

Let me know if it makes sense or not 🙂

AronHetLam, 2021-02-26:
Hope you got it working. I'll close this for now. If you need more help, feel free to add another comment. 😊

closed 2021-02-26 by AronHetLam