AronHetLam/ATEM_tally_light_with_ESP8266 #22: Led 2 for second ATEM input

frabusas, 2021-02-25:
Could LED2 be set to another ATEM input?
I would like to show the speaker which of two cameras is currently in the programme and would then only need a D1 mini.

AronHetLam, 2021-02-25:
Yes, but it's not possible change through the website (yet...), however, I could guide you to the code you'd need to change. It would require two places where you'd replace two words with a number instead.

frabusas, 2021-02-25:
great, can you tell me how and where I would have to adapt the code? Perhaps a screenshot?

AronHetLam, 2021-02-25:
Depending on if you want to change the first or the second LED you'd swap out the orange or the green ones. You'd swap `settings.tallyNo` with the input number minus one. So if you want it to show tally light for input 1 on the switcher, you'd put 0 (zero) in the parentheses.
[image]
You can search for `//set tally light` to quickly navigate to the relevant code be pressing `Ctr + f`.
If you only swap one of them you can still set the other one in the website, if you swap both, the settings from the website are ignored.

Good luck! 😊

frabusas, 2021-02-25:
Great, that works - thanks for the quick support

closed 2021-02-25 by AronHetLam