AronHetLam/ATEM_tally_light_with_ESP8266 #23: Neopixel brightness setting

goudi76, 2021-03-04:
Hy
could you please add a setup of the  brightness of for the neopixel?
as an Idea I have checked to do it.

I have done some changes in the code but I couldn´t arrange a needed setup in the .html part. 
In my code I have added these lines to setup the brightness and it works:
Line  70: int LEDBrightness = 100; //Value between 0-255
Line 164: FastLED.setBrightness(LEDBrightness);

It would be nice to have a setting line in the .html page for the lightness.
I have checked v3.0 yesterday with a Blackmagix 1/me and a 4k mixer and it worked perfekt! Thanks alot!

AronHetLam, 2021-03-07:
Try out this [commit](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/commit/99e28a93ed7ab6ec07c9dbaa8bc3b564da91cfc4), and let me know if that works as expected. Also please try it with a switcher for me, and let me know if i broke something else regarding the led strip. I made some small changes to increase performance/stability, and would like to be sure it didn't break anything before releasing it 😊

AronHetLam, 2021-03-07:
I found out I didn't know how to use memset(), so thins were definitely broken, a fix is again committed.

goudi76, 2021-03-12:
Thanks. What do you think will be a new version ready for testing?

AronHetLam, 2021-03-13:
I just put it into a new release yesterday, else, each commit made is downloadable from the main page.
Releases are just certain states of the project, that should be stable and tested versions, whereas the lates commit would be the bleeding edge version, with the latest changes. 
Hope that makes sense.

closed 2021-03-13 by AronHetLam