AronHetLam/ATEM_tally_light_with_ESP8266 #24: ATEM Connected but waiting

tspad1960, 2021-03-05:
I have the application up and running but do have an issue.  Once I have the network all configured and working, My status indicates  _"Connected - Wating for initialization - Connection might have been rejected"_  I see it is checking atemSwitcher.isConnected().  what could cause this condition?

Side info, I am running on ESP8266 NodeMCU

AronHetLam, 2021-03-06:
The connection flow is as such:
1. Tally sends hello packet to ATEM
2. ATEM responds to hello packet. (`IsConnected()` becomes true)
3. Tally acknowledges the response
4. ATEM sends initialisation data (`IsInitialized()` becomes true)

The tricky part is that the ATEM in step 2 could have rejected the connection, resulting in it never sending the initialization data in step 4. However, the library doesn't catch this, and just waits for step 4 (until a timeout, where it'll start over)

The ATEM only rejects connections if the connection limit is hit, so I'm wondering if you either have too many instances of software control running, while trying to connect a few of the tally lights?
I can't tell exactly what's going on in your situation, as I don't have enough details about switcher model, amount of tallys, if any of them connect and how many instances of software control you run etc.

goudi76, 2021-03-06:
Hy could you please show us your IP Setting? 
Only if it is a network not reachable from the Internet
You IP Adress
The IP Adress Mask
The Gateway IP
The Atem IP

Maybe there is already the failure.

tspad1960, 2021-03-06:
I am not on location with the ATEM at this time but the settings were

Device:
      IP: 10.0.10.50
     Mask: 255.255.255.0
     Gateway 10.0.10.1

ATEM
     IP: 10.0.10.30
     Mask  255.255.255.0
     Gateway  10.0.10.1

Thomas Spadaro

Thomas.Spadaro@SmartCommunications.us
Mobile: 609-509-0303

MarnitzRoux, 2021-03-07:
I had a look into doing this with an ESP32 a while a go and I ran into a similar issue, but upon further research, I found that the free, open source ATEM libraries by SKAARHOJ didn't support ATEM versions above 7.5. From my understanding that would mean anything that uses those libraries won't work with any more recent versions.

https://www.skaarhoj.com/fileadmin/BMDPROTOCOL.html

AronHetLam, 2021-03-07:
@MarnitzRoux that's not the case. I've had multiple people ask me the same thing, it just means that his piblic libraries aren't updated since 7.5, and that newer features aren't supported. I initially made this project last summer, using it with an ATEM mini pro (v8.2 or so). In my version of the library I've added support for knowing if the ATEM is streaming of not.

The ESP32 might not be supported, but I just got one home, and am planning to make it work soon.

MarnitzRoux, 2021-03-07:
@AronHetLam Yeah okay, that makes sense. I initially followed another project that implemented the libraries for the ESP32 so I thought that would work but I guess not. Hopefully you'll have more success.
I'll keep an eye on your project for any future support for he ESP32.

tspad1960, 2021-03-07:
Just to follow up, I will be testing this again this week when back on
location.  There may have been another issue going on with the ATEM.  It
may have been causing a problem.

tspad1960, 2021-03-08:
I was able to dig up an original gen 1ATEM television studio, set it up on
my local network and tally light connected right away.  So the issue I had
was in the network.  Sorry for the false alarm.  Currently using ESP8266
NodeMCU device.

Thanks for the good work.  If I make any significant changes I will do a PR.

AronHetLam, 2021-03-08:
No worries. I'll go ahead and close this then. 
What I don't get is that it connected, but never initialized. All other cases so far with network problems would cause it not to connect at all.
Usually the issue you initially described is caused by hitting the ATEM's connection limit. 

If you find out more about what exactly the cause was, please feel free to post it in here. 😊

closed 2021-03-08 by AronHetLam