AronHetLam/ATEM_tally_light_with_ESP8266 #25: ESP32 support

AronHetLam, 2021-03-07:
@AronHetLam Yeah okay, that makes sense. I initially followed another project that implemented the libraries for the ESP32 so I thought that would work but I guess not. Hopefully you'll have more success.
I'll keep an eye on your project for any future support for he ESP32.

_Originally posted by @MarnitzRoux in https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/issues/24#issuecomment-792244842_

AronHetLam, 2021-03-07:
I made a new issue for the subject, to keep the other one about the specific problem they're having.

@MarnitzRoux My version of the libraries support the ESP32, also my own TallyServer library, I just checked, and those works as expected. However, my Arduino code doesn't yet, as WiFi things and pins works just differently enough to break it.

I'm looking on to what needs to change. 😊

AronHetLam, 2021-03-07:
This [commit](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/commit/ee20e73562cb8b886e5ab89f32469f875dff3dd8) makes it work on the ESP32 in my test setup, with an ESP8266 as a "server". It's not tested with an ATEM, but There shouldn't really be a difference. 😊

Depending on what ESP32 you have you need to change what pins you use.

MarnitzRoux, 2021-03-07:
Oh awesome, this sounds promising. I have the ESP32 PICO D4 board board for my testing so far but I'm planning on using the Beetle ESP32 which has the WROOM processor, which seems to be pretty common so I'm assuming that'll work.

I won't have access to the ATEM till the end of the week so I can't update if it worked or not for a bit but I'll let you know when I have tested it.

AronHetLam, 2021-03-08:
I did my testing on a DOIT DEVKIT V1, which is based on a WROOM as well.

MarnitzRoux, 2021-03-12:
So I just got a chance to test the tally with my board and an ATEM and it works perfectly. Just what I was looking for.

Thanks for your help :)

AronHetLam, 2021-03-12:
Sound good. I'll push it into a release soon 😊

closed 2021-03-12 by AronHetLam