AronHetLam/ATEM_tally_light_with_ESP8266 #26: Case for neopixel

goudi76, 2021-03-12:
Hy @AronHetLam,

I have created a case with the LED Stripe inside.
please have a look to this link and open the .pdf [file](https://drive.google.com/file/d/1VnltmGeJF-X3faQNYa1_LgQ1o2Slr4XT/view?usp=sharing)

[image]

I hope you like it and you will put it to your project page.
Right now I have two of them running in a house of worship - but I´m looking to install more - but 3D printing takes time

AronHetLam, 2021-03-14:
Looks good 😊
Wish I had a 3d printer...

But I'd have to think about how to include it on my page, as you're not the only one having made a case.
For now I'd recommend you to upload all the necessary files in your fork of the project, so that people can find them all there. Then I could make a collection of links to your and other's forks.

Hitower78, 2021-03-14:
Nice! 
My Case look more like a Block :-) Base is the DNorm Neutrik design, 
Wemos D1 inside with 2nd PCB on Top and 3 LEDs (Red, Green and rgb)
Last Times i use a 1/4" Threaded Inserts, but there are not available in Germany. I Will try to print 3/8.

When it is ready i will sho it here, because your Project is really fine :-)

PS: everyone needs a 3D printer... :-P

goudi76, 2021-03-15:
Hy I have set up the files in the fork.
I´m not that educated with GitHub and I don't know if you can bring the files together?
BR
Tomorrow I will test the "on air" light with a real ATEM switcher - using version V3.1

AronHetLam, 2021-03-23:
I made a page in the wiki where I'll link to custom cases, else this repo will become a mess! 
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/wiki/DIY-cases

closed 2021-03-23 by AronHetLam