AronHetLam/ATEM_tally_light_with_ESP8266 #28: Cannot change tally light name and set static IP

fareign, 2021-03-15:
If tally light name have space ,cannt change it.

static ip I chanage the code ,it's OK now

`    Serial.println(settings.tallyName);
    //Serial.println(settings.tallyIP[0]); //Check size of settings struct
    IPAddress local_IP(settings.tallyIP[0], settings.tallyIP[1], settings.tallyIP[2], settings.tallyIP[3]);
    IPAddress gateway(settings.tallyGateway[0], settings.tallyGateway[1], settings.tallyGateway[2], settings.tallyGateway[3]);
    IPAddress subnet(settings.tallySubnetMask[0], settings.tallySubnetMask[1], settings.tallySubnetMask[2], settings.tallySubnetMask[3]);
    //Serial.println(local_IP); 
    if (settings.staticIP) {
        WiFi.config(local_IP, gateway,subnet);
    }`

AronHetLam, 2021-03-15:
I can't reproduce the issues you're describing. I can give it names with spaces, and the static IP seems to work fine in my case.

This is loaded after it's saved and restarted:
[image]

What version are you using? Try to re-download the latest version, as I made a fix yesterday. The picture is from the v3.1 that's there now.

fareign, 2021-03-16:
thanks ,its ok now.

closed 2021-03-16 by AronHetLam