AronHetLam/ATEM_tally_light_with_ESP8266 #30: i am confused: my Red and Green is reversed

Hitower78, 2021-03-16:
#Hi there

after setup my wlan (the other issue) am am working on a new case.

Something wicked happen: 
my green and RED LED is reversed. 
The Code is the latest (incl Ws2812 support), Here is the define section

#define PIN_RED1    D0
#define PIN_GREEN1  D2
#define PIN_BLUE1   D1

//Define LED2 color pins
#define PIN_RED2    D4
#define PIN_GREEN2  D5
#define PIN_BLUE2   D6

But: At my Wemos D1 from AZ Delivery this happen: 
Red LED is D2, Green LED is D0. 
First i think i solder the wrong LED, but when i use 5V directly everything ist OK (the green LED is Green)

D4 (2nd LED Red) is very low and always on. This happen to every Wemos D1 in my setup (different settings, breadboard...)

WS2812b is working fine! Thanks :-)

I didnt understand that :-(

its no problem, because i can change the code or the Hardware, but... i didnt understand that

goudi76, 2021-03-16:
Well, that is a problem of the different supplier and coding of ws2812b stripes.
The Chanel for each color is coded but the direction is not fixed. Most common is GRB (green, red, blue) or the other way around RGB (red, green, blue).
The blue light should be blue in any case...

change the code
` 178       leds = new CRGB[settings.neopixelsAmount];`
to
`178       leds = new CGRB[settings.neopixelsAmount];`
If this does not work change also the direction in lines 103-105 to `CGRB`

@AronHetLam It would be good to bring the direction GRB and RGB up to the Web Page as a set point as well.

Hitower78, 2021-03-16:
Hi There,
i didnt use the ws2812 in this setup. 
i use 10mm RED and Green LED and 5mm RGB (2nd)

I notice this RGB and GRB direction "Problem" because i am using different WS2812 Stripes in a JINX setup.
The Direction in the Setup would be nice.

Hitower78, 2021-03-16:
[image]
[image]

Here you can see the glowing Red.

In teh 2nd Picture you can see: 
Yellow is RED 10mm LED on D2
Red is RGB LED R
Blue is GND
Blue (left) is RGB Green on D5
Violett is RGB Blue on D6

AronHetLam, 2021-03-16:
What board did you select?
Sounds like issues caused by selecting `WeMos D1 R1 (esp8266)`, and not `LOLIN(WEMOS) D1 R2 & mini (esp8266)`
The pinout is different on different boards, so it wont match if you have the wrong one. 😊

[image]

Hitower78, 2021-03-16:
uff. thats a good question :-P

I will try it with the LOLIN one :-)

thanks :-)

closed 2021-03-16 by Hitower78

AronHetLam, 2021-03-16:
> Sometimes the Problem ist 50Centimeter in Front of this Screen!

I know that feeling! 😂