AronHetLam/ATEM_tally_light_with_ESP8266 #33: Won't compile

bruce1946, 2021-03-19:
This is the message I get from the IDE:

ATEM_tally_light:261:22: error: 'class WiFiClass' has no member named 'mode'

                 WiFi.mode(WIFI_STA); // Disable softAP if connection is successful

Any idea what went wrong?
Thanks for your help.
I am not sure what the rest of the stuff below here is.

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, WIFI, Only Sketch, 921600"

In file included from C:\Users\HPz800\Documents\Arduino\ATEM_tally_light_with_ESP8266-3.1\ATEM_tally_light\ATEM_tally_light.ino:50:0:

C:\Users\HPz800\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^

In file included from C:\Users\HPz800\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,

                 from C:\Users\HPz800\Documents\Arduino\ATEM_tally_light_with_ESP8266-3.1\ATEM_tally_light\ATEM_tally_light.ino:50:

C:\Users\HPz800\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from C:\Users\HPz800\Documents\Arduino\ATEM_tally_light_with_ESP8266-3.1\ATEM_tally_light\ATEM_tally_light.ino:43:0:

C:\Users\HPz800\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h: In instantiation of 'class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>':

C:\Users\HPz800\Documents\Arduino\ATEM_tally_light_with_ESP8266-3.1\ATEM_tally_light\ATEM_tally_light.ino:126:24:   required from here

C:\Users\HPz800\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:53: error: no type named 'ClientType' in 'class WiFiServer'

   using ClientType = typename ServerType::ClientType;

                                                     ^

C:\Users\HPz800\Documents\Arduino\ATEM_tally_light_with_ESP8266-3.1\ATEM_tally_light\ATEM_tally_light.ino: In function 'void setup()':

ATEM_tally_light:228:10: error: 'class WiFiClass' has no member named 'mode'

     WiFi.mode(WIFI_STA);

          ^

ATEM_tally_light:228:15: error: 'WIFI_STA' was not declared in this scope

     WiFi.mode(WIFI_STA);

               ^

ATEM_tally_light:229:16: error: no matching function for call to 'WiFiClass::begin()'

     WiFi.begin();

                ^

C:\Users\HPz800\Documents\Arduino\ATEM_tally_light_with_ESP8266-3.1\ATEM_tally_light\ATEM_tally_light.ino:229:16: note: candidates are:

In file included from C:\Users\HPz800\Documents\Arduino\ATEM_tally_light_with_ESP8266-3.1\ATEM_tally_light\ATEM_tally_light.ino:5:0:

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi\src/WiFi.h:60:9: note: int WiFiClass::begin(char*)

     int begin(char* ssid);

[... 208 lines cut ...]
C:\Users\HPz800\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:91:3: error: 'class WiFiServer' has no member named 'close'

   _server.close();

   ^

Multiple libraries were found for "EthernetUdp.h"

 Used: C:\Users\HPz800\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\Ethernet

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Ethernet

Multiple libraries were found for "WiFiClient.h"

 Used: C:\Users\HPz800\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

'class WiFiClass' has no member named 'mode'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

AronHetLam, 2021-03-20:
That's a first... I'm not sure why, but it seems it's getting hold of the wrong WiFi library, that doesn't have the features it needs. Make sure it gets the ESP8266 one.
Try turning on verbose mode in settings, and post what it the outputs. To make it look nicer, put it all in ` ```  post  ``` `
[image]
Then I may be able to see what files exactly it's trying to use.

If you've changed your include statements, you could post those as well.

AronHetLam, 2021-03-20:
I would assume you would get the same problems running ESP8266 examples using WiFi

AronHetLam, 2021-04-02:
Any  update on this issue?

AronHetLam, 2021-04-17:
I'll close this as it's been inactive for a while now.

closed 2021-04-17 by AronHetLam