AronHetLam/ATEM_tally_light_with_ESP8266 #34: Using ATEM_tally_light_with_ESP8266 within obs

Stephan-Kl, 2021-03-24:
Hey,
I'm currently working on the tally lights for my ATEM mini pro ant its very straight forward and simple to assemble.
Is it possible with this setup to get the OBS Tally information via web sockets implemented? Maybe that one LED in the Neo-Strip change to red when OBS ist streaming?

kind regards
Stephan

AronHetLam, 2021-03-25:
Glad to hear you like it, and find it easy to make. 
There is a websocket library for Arduino, that should work on ESP as well. I don't know about the OBS part, but if it has a websocket to connect to It would be possible. But WiFi performance might become a problem.

Stephan-Kl, 2021-03-25:
Ah okay. My programming skills are not as good to implement these by my self :D