AronHetLam/ATEM_tally_light_with_ESP8266 #35: If disconnect wifi ,can not auto connect.

fareign, 2021-03-28:
Today,I test the distance how long it can be connect.
I walk a long distance,It's always a pink light, actually  it's  disconnected.

AronHetLam, 2021-03-28:
Interesting issue. I think I know where to look.

Do you know it connects to WiFi again once back in range, and just doesn't reconnect to the ATDM, or if it doesn't reconnect to WiFi at all?

goudi76, 2021-03-28:
I have Seen something similar. IZ connects to the WLAN but not to the Atem.
Sometimes it conects after a while.

AronHetLam, 2021-03-28:
Just for laughs, something up with time zones... Or git can now tell the future 😂 
[image]

AronHetLam, 2021-03-28:
Problem was that when connecting to ATEM it would stop checking if the WiFi connection is lost or not, which could make it get stuck in the connecting to ATEM state. I made the WiFi check run always, so now it should be able to see if WiFi is lost, no matter the state.

I'm not sure why it wouldn't connect to the ATEM automatically once it got WiFi again, even though it didn't show the no WiFi state, because the WiFi reconnection happens automatically, outside the Arduino code (and other parts of my code). If any of you still have that issue, please let me know, and I'll try to look more into it. But I think that should be fine now as well, as now the ATEM connection gets reset properly after WiFi is back. It didn't before, and i suspect that would cause it to break.

goudi76, 2021-03-28:
That is exactly the main prob. Always I have the computer connected and the Serial monitor is open, everything works. If the ESP is allone I have seen such effects. But I do not know if this might be a problem with the router or in the software of the Tally light.
Otherwise I would have posted a record, but I couldn't ever grep this. The main Problem is, that I haven't found a way to reproduce this problem. Sometimes it connects - sometimes it does not. @fareign: Is it always the case that it doesn't find the Atem after a loss of the WLan?

fareign, 2021-03-28:
I test a long time . ESP8266 and ESP32 have been used together to do the test.
They all have lost connection from ATEM(one hour about 2-3times),allmost at the sametime .sometimes  can be reconnect and sometimes cannot reconnect,must reset ESP. so I think ,the condition about not reconnect is disconnect to WIFI.

AronHetLam, 2021-03-29:
The WiFi related part should work now, but the periodical connection loss to the ATEM I assume isn't a connection loss to WiFi also. Or do you know if those connection losses where related to WiFi as well?

The WiFi problem I solved in commit 7b33888 has been there from the start, and I've never had issues with the ATEM connection, unless i also had WiFi problems. My production units run for hours without problems, but they have a version from before Neopixel support was added. The periodical disconnect (at the same time for units I assume turned on at the same time.) could be caused by the FastLed library, as it messes with interrupts, that are used for tracking time... That might be a harder one to solve...

fareign, 2021-03-29:
thanks ,I try to cut fastled and the second led, and do a test.

AronHetLam, 2021-03-29:
@fareign 2nd led won't have an impact, that code is on my units as well.

fareign, 2021-03-31:
when I cut Neopixel code ,they works very well. no longer loss connection. THANKS!!

closed 2021-04-02 by AronHetLam

Andrej65416das, 2021-04-08:
@fareign Could you please send me exactly which part should be cut out?

Thanks