AronHetLam/ATEM_tally_light_with_ESP8266 #36: Support for ATEMs with more than 2 M/Es

MarnitzRoux, 2021-03-29:
So I've been running the tally program for the last couple weeks on an ATEM TV Studio Pro and it works perfectly.

I wanted to show it to a friend, who works in production, on their ATEM at work but they have a 4 M/E switcher and the SKAARHOJ libraries only support up to 2 M/Es (identified cases from line 84-108 in the ATEMmin.cpp file). I'm assuming that was done because the 4 M/E didn't exist when the libraries were made.

Even though you didn't create those libraries, do you think you could have a look to see if a mod could be made to support up to the 4 M/Es?

AronHetLam, 2021-03-30:
For receiving tally data what's in line 84-108 doesn't really matter, as it's not used at all. The only part I use is in line 314-143, which gets the tally data. I did once raise the limit from 20 tally inputs to 40, to make it support the latest models.
Did you try it out on the 4 M/E?

you could try to add `Serial.println(sources);` after line 316 to see what value `sources` have. If it's more than 40 I need to raise it again.

MarnitzRoux, 2021-03-30:
Yes we did try it with a 4 M/E but it didn't respond to any input from the switcher itself: it connected to the network and connected to the switcher but never showed any tally response on any of the M/Es.

I'll add the print sources and see try it on the 4 M/E again, it won't be for a few weeks though.

AronHetLam, 2021-03-30:
Ok. What model is it exactly?

I found this in the manual for the 8K 4 M/E version
[image]

It's not for the UDP protocol (that's used over networking) but for their camera protocol. But the max amount of sources should be the same, so maybe I should raise it to 510 to be save. 😅

MarnitzRoux, 2021-03-30:
Well that's a heck ton of devices. We tested it on the 4K 4 M/E with 20 inputs. But this is talking about sources, not just inputs so I can believe that it would be much more than the input number.

I suppose it would cover all the bases if it was set to 510? So might as well and see what happens.

AronHetLam, 2021-04-02:
I raised the limit to 510 in this new [branch](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/sources-limit). The tally server won't work with more than 40 sources for now.

It could also be something else. I stumbled over a comment in the ATEMbase lib saying that they observed up to 32 init packets (which would vary) on a 4k 2m/e, so they set the limit to 40 init packets to be safe. I have no clue if it would be more and how much more on the 4m/e. The only way to find out would be by doing a few captures using wire shark... which is mainly how the protocol has been reverse engineered.