AronHetLam/ATEM_tally_light_with_ESP8266 #37: Possibility to Adress Tally with Hardware

its-Blackpoint, 2021-03-31:
For a changing Enviroment it would be nice to address the Tally by Hardware.
For example with 5 Dipswitches in the  Style that DMX Lighting Fixtures were addressed. 
That would give the Possibility to Change 31 Inputs verx Quickly.

AronHetLam, 2021-04-02:
I don't have a m5stick-c or Atom matrix myself, but had a similar idea to display tally number on the display, and have the button scroll through tally numbers. They are based on the ESP32 chip, så my code should work on them. Only the display and button code needs to be added.

falcaobh, 2021-06-30:
Any news about this?

AronHetLam, 2021-07-05:
Not much as I as mentioned don't have a m5stack unit with a display.

On the D1 mini the option with dip-swithces isn't possible, as there aren't enough pins left attach them to, and that's besides having to somehow make the use of it optional.
Also as it doesn't have a display, scrolling through numbers with a button press is a bad guessing game. I made a new branch with the possibility to add a button to pin D3, which will count up the tally number for each press (short to ground).
I don't think it will be pleasent to use, but you can try it out. Also, at the moment it doesn't save the changes on button presses, so it resets when rebooted.

falcaobh, 2021-07-05:
You can send me the code?

Enviado do meu iPhone

AronHetLam, 2021-07-06:
Here's a link to the branch https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/tallyNo_btn

Click the green `Code` button and select "Download ZIP"
[image]

falcaobh, 2021-07-06:
Thank you from Brazil!!

falcaobh, 2021-07-07:
Thanks for the previous help.
> Using the d3 button worked perfectly.
> I would like to know how to find the tally number that I selected in the
> web interface to show on the serial monitor so I can show it on an oled
> display. I found tallyNo; but I don't know how to use
> Serial.println("xxxxxxxx " + ?(tally number)?().toString()); .
> Can you help me? when it's ready my code with OLED and button I'll give
> you due credit.
>

AronHetLam, 2021-07-07:
Seems like somethings missing from your email here on GitHub. It's better if you post the comments on GitHub directly 🙂

dmitrylive, 2023-06-02:
I have been trying to learn arduino for a very long time... and on the original board, the pin could be both an input and an output. I could be wrong. Contacts D6, D7, D8 are free. could they be used? There is nothing wrong with a DIP switch, it is still used on lighting fixtures today. only there are 512 channels, and nothing, everyone figured out where to click))

AronHetLam, 2023-06-02:
Correct, the pins can be both in- and output, but only pin D3 and D8 are free, and D3 is even used by the push-button solution presented in this issue. 
D4, D5 and D6 are used for a second LED, and D7 is used for addressable LED like Neopixels. 

To cover the 41 possible input channels some ATEMs have, 6 dip-switches are needed. So sadly it's not possible to make a general enough solution with the use of dipswitches.

But since you've worked with Arduino before, you can modify it for your needs. 

You can use the following snippet (not tested). It reads each of the pins and converts it into a single number by left-shifting (`<<`) the values, and overwrites whats read from the settings. Just make sure it comes after `EEPROM.get(0, settings);` in `setup()`.

``` C++
int dipTallyNo = digitalRead(dip_1_pin);
dipTallyNo = dipTallyNo | (digitalRead(dip_2_pin) << 1);
dipTallyNo = dipTallyNo | (digitalRead(dip_3_pin) << 2);
settings.tallyNo = dipTallyNo;
```

The simplest would probably be to change the pins used for LED 2 to INPUT, and comment out  lines with `setLED(color, PIN_RED2, PIN_GREEN2, PIN_BLUE2);` so that they're not written to. Also use 3.3v, as the inputs can't officially handle 5v.

I hope that gets you going :)