AronHetLam/ATEM_tally_light_with_ESP8266 #41: Neopixel Control

MarnitzRoux, 2021-04-20:
I'm wanting to use two 8mm PTH Neopixel LEDs for my tally, one facing forward to the speaker and one facing backwards towards the camera operator. I know it would be possible to send the data signal to both LEDs and just have them mimic each other, however, I want to have a lower brightness for the camera operator since the LEDs are quite bright and the operator is much closer than the speaker.

This means I'll need two data signals, which seems to be provisioned for with a second Neopixel LED data pin, but it's unclear what exactly this second LED does; whether it's program and preview like the regular two LED setup, if it just duplicates the signal two both or if LED1 does program and preview but LED2 only does program in the case of a front facing LED for a speaker to only see which camera is live.

I've gone through the Arduino code and can see where it controls the Neopixels but the method it uses is unfamiliar to me.

TL;DR: How exactly does having two Neopixel LEDs work? Is there a way to control the brightness of the two LEDs independently?

AronHetLam, 2021-04-20:
Currently no, there's only an overall brightness control for all neopixels at once, but it's possible to implement.
You could hardcode it in your version, that would be the simplest for now.
It sounds like you ably want two neopixels, and none of them to act as a status led. I'll post a line of code and where to put it later today if you need it 🙂

MarnitzRoux, 2021-04-20:
Yeah, similar to the functionality of the regular LEDs where you can choose the modes, where you can choose to show program, preview, or both.
The argument can be made that if the LED facing the speaker is only showing program, I can just use a regular red LED but I want to keep that functionality for future use-cases. And because the only 8mm LEDs I have are Neopixels.

I think I've figure it out though, I can modify the `setSTRIP` function to `setNEO1` to only apply to the first Neopixel and make a second `setNEO2` function to control the second Neopixel and set them for different LED modes so they can function differently on the same data pin. I'll only be able to test that on Friday though.

AronHetLam, 2021-04-20:
Cool. Something like that would be my quick fix as well.

closed 2021-04-20 by AronHetLam