AronHetLam/ATEM_tally_light_with_ESP8266 #44: How to connect more Tally

xx1ong, 2021-05-20:
Hello author, first of all thank you for your dedication to how good teaching.

I now have 10 tally following your tutorial

Only 4 will be able to connect to ATEM for use after tally is set

The other 6 cannot be connected (LED is pink)

See you say version 2.0 has added the "Tally Server" feature

This will enable more entrants to join ATEM

But I can't seem to find this feature

Could you tell me where this function is set

Or what methods can be used to link more numbers

I am looking forward to your answer. Thank you

AronHetLam, 2021-05-20:
Glad to hear you can use the project.

The tally server feature lets you connect one tally to another one already connected to an ATEM (or another tally). The way you do it is by setting the "Switcher IP" to the IP address of another tally already connected, instead of the IP of the ATEM.

Did you see the illustration in the README? [README](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266#what-does-it-do)

xx1ong, 2021-05-20:
Thank you very much for your reply and guidance, and thank you for your long-term support in this work!

Wish you a happy life!

AronHetLam, 2021-05-20:
Thanks, i appreciate it 😊

closed 2021-05-20 by AronHetLam