AronHetLam/ATEM_tally_light_with_ESP8266 #45: WI-fi connection

estracke, 2021-05-21:
Hi, thanks for this very usefull tally system. I bought a Nodemcu D1 Mini Esp8266 Wifi arduino and did the steps in tutorial. Connected in a Tally Light Setup network and configured all items in question. But the connection with my wireless network is not working. Have any conditions to a SSID network name or a security protocol like WEP, WPA etc? How can i see the logs or debug to know where is the problem?

Thank very much!

AronHetLam, 2021-05-21:
How far did you get? I assume you uploaded the firmware and is unable to get to the setup page.
If that's the case, did you connect to the `Tally light setup` WiFi? Once you connect to that WiFi you can access the setup page by typing `192.168.4.1` In the address bar of your webbrowser.

estracke, 2021-05-24:
Hi Aron, thanks for your reply. I´m just at the point you mentioned, I can connect setup page but when i put my confs in SSID, password, IPs, etc and save, but message still on changing state..... and atem connection is disconnected, waiting for wifi

estracke, 2021-05-24:
Hi... i used serial monitor to log messages and that is the output:

Serial started
LiveFactory
------------------------
Connecting to WiFi...
Network name (SSID): 
Unable to connect. Serving "Tally Light setup" WiFi for configuration, while still trying to connect...

Seems like is not saving SSID network when i put in configurations page. There is a way to force this SSID name in source code, just to test?

Regards

estracke, 2021-05-24:
New try.... when i forced WiFi.begin("LFTALLY","********"); it works like below... seems like is not saving parameters in 192.168.4.1 browser page.

########################
Serial started
⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮
------------------------
Connecting to WiFi...
Network name (SSID): LFTALLY
------------------------
Connected to WiFi:   LFTALLY
IP:                  (IP unset)
Subnet Mask:         (IP unset)
Gateway IP:          (IP unset)
------------------------
Connecting to switcher...
Switcher IP:         255.255.255.255

estracke, 2021-05-24:
I did all tests... it seems that 192.168.4.1 don´t commit ONLY SSID and PASSWD... everything else is commit normally:

Serial started
LFTALLY#1
------------------------
Connecting to WiFi...
Network name (SSID): LFTALLY
------------------------
Connected to WiFi:   LFTALLY
IP:                  10.10.2.103
Subnet Mask:         255.255.255.0
Gateway IP:          10.10.2.253
------------------------
Connecting to switcher...
Switcher IP:         10.10.2.241

AronHetLam, 2021-05-25:
Seems like you know how to make the workaround, and that you aren't stuck because of it.
But it needs a looking into. 

With what version is this?

Also, if you wrap the output like ` ```  OUTPUT ``` `, then it will format nicely :)

estracke, 2021-05-26:
Hi Aron, as i am not an expert, i have beem reading the code and relized that ssid and pwd was not being stored on EEPROM and code WiFi.begin() was not assumed parameters. Then i brought this parameters in a char[] array and stored in EEPROM and i placed new parameters like WiFi.begin(settings.ssid, settings.pass) and works fine. 

Thanks for your code... it helps me very much.

AronHetLam, 2021-05-26:
@estracke What release version is this?

AronHetLam, 2021-05-26:
The SSID and key for the WiFi isn't stored by my code, while the WiFi library does this by default.

Take a look at this [line](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/ea23fa9122ae4b3216ef9338e013aea29dc72146/ATEM_tally_light/ATEM_tally_light.ino#L746). Here the new WiFi credentials are set, and should be stored as well.

estracke, 2021-05-26:
How do i know the version? 

I´ve just declared in struct settings:

char ssid[32] = "";
char pass[32] = "";

Stored:

val.toCharArray(settings.ssid, (uint8_t)32);
val.toCharArray(settings.pass, (uint8_t)32);

and connect:

WiFi.begin(settings.ssid, settings.pass);

worked fine!

AronHetLam, 2021-05-26:
The version you've downloaded. Latest release being  `v3.1.1`.

Just tested with my phone over the setup network it serves. Works fine for me.
Here's my output:
```
########################
Serial started
ESP-7941FC
------------------------
Connecting to WiFi...
Network name (SSID): Test ssid
Unable to connect. Serving "Tally Light setup" WiFi for configuration, while still trying to connect...
```

Here i change the SSID over the setup page, and the chip resets:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
########################
Serial started
ESP-7941FC
------------------------
Connecting to WiFi...
Network name (SSID): New test ssid
Unable to connect. Serving "Tally Light setup" WiFi for configuration, while still trying to connect...
```

What you've added shouldn't be needed. Have you selected the correct board and settings in the board manager, sometimes selecting the wrong one can cause weird issues.

estracke, 2021-05-26:
Hi,

    this is my board: [https://produto.mercadolivre.com.br/MLB-776074297-modulo-nodemcu-d1-mini-esp8266-wifi-iot-arduino-0083-_JM](url) and my selected board is LOLIN (VEMOS) D1 R2 & mini.

Thanks,

commit 1722d6ec (2021-06-19): Fix #45 & #47 - WiFi credentials not being persisted after ESP lib update
closed 2021-06-19 by AronHetLam