AronHetLam/ATEM_tally_light_with_ESP8266 #46: Tally Server Problems setting up the

xx1ong, 2021-05-23:
Hello author, I have a question about the connection between the tally lamps
My ATEM can add no more than 4 TALLY
You can't add more than 4
Then I remember you saying that version 2.0 has added the "Tally Server" feature
I set the newly added Tally, switcher IP to one of the normally working Tally static IPs
The newly added Tally still shows that pink lights are not connected and not working properly
Below is my setting, please help me to have a look
Thank you very much for taking the time to help me out

**Main information:**
ATEM IP ：192.168.10.240
gateway：192.168.10.250
subnet mask：255.255.255.0

1.tally3(Use in normal connection）

TALLY IP：192.168.100.30
subnet mask：255.255.255.0		
gateway：192.168.10.250
ATEM IP：192.168.10.240

————————————————

2.tally4(Use in normal connection）

TALLY IP：192.168.100.40
subnet mask：255.255.255.0		
gateway：192.168.10.250
ATEM IP：192.168.10.240

————————————————

3.tally5(Use in normal connection）

TALLY IP：192.168.100.50		
subnet mask：255.255.255.0		
gateway：192.168.10.250
ATEM IP：192.168.10.240

————————————————

4.tally6(Use in normal connection）

TALLY IP：192.168.100.60			
subnet mask：255.255.255.0		
gateway：192.168.10.250
ATEM IP：192.168.10.240

————————————————

**5.tally2(New) Unable to connect**

TALLY IP：192.168.100.20		
subnet mask：255.255.255.0		
gateway：192.168.10.250
ATEM IP：192.168.100.30

AronHetLam, 2021-05-25:
The one thing i notice could cause this is that you are running on two different subnets. I'm not a network expert, but as far as i know having a subnet mask that looks like `255.255.255.0`, results in
`192.168.10.xxx` and
`192.168.100.xxx`
being on each their subnet, making them unable to connect to each other. Why they can connect to the ATEM i don't know, maybe the subnet mask on you router and/or ATEM are set differently?
However, I would think keeping your tally IPs in the range of `192.168.10.xxx` would make it work.

Another thing i note is that I've never seen a gateway IP ending with `250`. It could easily be valid, but are you sure that that's the actual gateway IP on your network?

If not, let me know - but right now i can't think of anything else causing this issue.

xx1ong, 2021-05-28:
Thank you very much for your reply
I will try again today according to the method you said
I will continue to update the solution in this article

xx1ong, 2021-06-17:
Thank you for your reply in your busy schedule
I've solved the problem
The problem is as you said IP segment XXX.XXX.100.XX
Thank you

AronHetLam, 2021-06-19:
Glad top help! 😊

closed 2021-06-19 by AronHetLam