AronHetLam/ATEM_tally_light_with_ESP8266 #47: WiFi settings seem not to be soring on ESP Board

bruce1946, 2021-06-17:
Hi Aron:
I made a few of these tally modules and am having troubles. It seems that the WiFi settings (SSID and password) are not being stored on the ESP8266 board. I am using version 3.1.1 of your software. I see that others are having this problem as well. Do you have any suggestions to correct this? Or, perhaps, you are working on a new version to fix this? Thank you so much for your help.

xx1ong, 2021-06-17:
I've had the same problem

3.0 shows that WiFi and password cannot be saved after successful saving

The problem appears to have arisen after the Arduino library files were updated

I still couldn't fix the problem and eventually went back to version 2.0

AronHetLam, 2021-06-19:
Didn't know there was a new version of the ESP Arduino libraries. Seems like those changed the way they used to persist WiFi credentials automatically.

That last commit should fix it, at least it does for me. 😊
Let me know if it does for you to, and I'll release the patch for it.

closed 2021-06-19 by AronHetLam

bruce1946, 2021-06-19:
Hi Aron:
Yes!!! it worked. Thank you so much for fixing this.
Bruce Goguen

AronHetLam, 2021-06-19:
You're welcome 🙂