AronHetLam/ATEM_tally_light_with_ESP8266 #51: Tally light no red LED

aexeler, 2021-08-17:
Hi,
I'll try out your tally light (great job!).
I flash a D1mini (with RGB-LED) with the ATEM_tally_light.ino, open the config page (192.168.4.1) and set the SSID. PASS, tally light number (1) and the IP of my ATEM (is an ATEM Mini Extreme ISO).
The LED turns green, but when I go online with ATEM Channel 1, the LED does not turn red.
Any idea what I'm doing wrong?

Greets and thanks for help
Andreas

AronHetLam, 2021-08-19:
Looks like you have done the software and setup part correctly, so my guess is it's a hardware issue. To try it you can use the Tally_test_server.ino which will let you cycle through the different tally states via the serial monitor in the Arduino IDE. If the LED colours are still incorrect there, it's probably bacuse the led Is mounted incorretctly, or maybe it's common anode led, and not common cathode?

AronHetLam, 2021-08-19:
The latest release doesn't contain the latest version of the Tally_test_server.ino, but you can download that directly from the main GitHub page. It fixes a bug with saving WiFi credentials.

aexeler, 2021-08-19:
Hi,
OK, i´ll test it with the Tally_test_server.ino today.
Thanks for help!
Greets
Andreas

aexeler, 2021-08-28:
Hi,
with the Test_server.ino, i cant change the LED-Color.
So i flash the Tally_light again with clear config and wlan settings.
Now it works!!!
Maybe i played to much with the Settings, LED_Mode, RGB and Neopixel, Wlan-setting and so on.
Thanks for help
Greets Andreas

AronHetLam, 2021-08-28:
Strange, but good that it works. 🙂

closed 2021-08-28 by AronHetLam