AronHetLam/ATEM_tally_light_with_ESP8266 #53: Which WIFI router?

jmvb-cl, 2021-10-17:
Hello
My setup is now 5 tally light

I have an ATEM  1 M/E production studio 4K and an ATEM 1 M/E broadcast panel. I don't use a computer for control
My first router was a TP-LINK TL-WR740N
Since I had range and drop issues, I thought changing it for powerfull one, the problems will gone.
I get a TP-LINK TL-WR940N, but the issues remain, every 30 or 40 camera change, got frozen the control panel for a few seconds. Sometimes all tally's lost connection, sometimes not.

Doing some research, I found that the control protocol uses UDP and is fairly intolerant to packet loss and network latency and as Strata pro app developers suggestion, now I'm trying an Apple Airport express.
In home test, at least the 40 changes freeze problem gone, but I concerned about the router range, since it lacks of external antenas. I hope do some in field test soon.

Which router are you using? do you have drops problem?

A guy form youtube recommend a D Link Dir 615, someone have use it?

jmvb-cl, 2021-11-02:
Well, did 4 other jobs in a busy street in downtown, and was a disaster.
The tallys lost the signal after a few meter of both routers

I borrow a Mercusys AC12, 5 x 5dbi antennas and obtain better results. 
Fixed cameras never lost the signal, and the mobile one, works ok 90% of time.

jmvb-cl, 2021-11-25:
well Mercusys have issues too.
Last 2 job, in a indoor Basketball Court,  first day, drops every 5 or 10 minutes.
Second day, I put the router in a mic stand in height, far from Atem, and perform better, only a couple of drop.

Just bought a Ubiquiti unifi UAP-LR and despite I only did home test, seems like I have a jackpot
No drops or freeze for at least 2 hour test

jmvb-cl, 2021-12-04:
Ubiquiti unifi UAP-LR is the best so far.
I did several jobs and only a few drops per event.
The Atem freezes (if they occur) only last a blink (instead at least a second with the previous routers)

AronHetLam, 2021-12-04:
Recently I tried updating our ATEM mini pro to the latest version, which resulted in it freezing sometimes in the middle of a show. We feed the video to OBS over usb, and assumed it was an issue with the usb connection in the latest firmware. We downgraded the ATEM again, and haven't had issues since.

It just struck me that it might be that the latest ATEM firmware is more sensitive to data loss/slow response or something like that, and that it was actually my tally lights causing the freezing.
I don't have time to look more into if that is actually what was causing the freezes for us, but I can look up what stem version I downgraded to of want to try it out.

jmvb-cl, 2021-12-13:
Interesting... 
My atem control software is 8.6.4... which one is yours?

bfbones, 2021-12-13:
Be careful while downgrading those ATEMs, already bricked two of them trying to downgrade. Also no chance using recovery then.
 
 I also experience drops since some weeks, I also upgraded my ATEM Extreme some time ago - maybe this is really a firmware problem?
 I am using 4 Tallys.. maybe there are already too much connections?

jmvb-cl, 2021-12-13:
> Be careful while downgrading those ATEMs, already bricked two of them trying to downgrade. Also no chance using recovery then.
>
> I also experience drops since some weeks, I also upgraded my ATEM [...]

Before upgrade, had you drop issues?

AronHetLam, 2021-12-13:
> Be careful while downgrading those ATEMs, already bricked two of them trying to downgrade. Also no chance using recovery then.
>
> I also experience drops since some weeks, I also upgraded my ATEM [...]

Sounds like no fun! 🙁

Might sound like we're running into issues with the ATEM libraries by Skaarhøj not being updated as the ATEMs firmware is. If that's the case I sadly don't have the time or resources to keep it up to date.

jmvb-cl, 2021-12-14:
Well, for me, the system which I build with 5 tallys and the Ubiquiti AP work very fine.
I can live with a couple of drops per night

Good work!

jmvb-cl, 2022-01-25:
Any of you guys know if an "Isolate" network for the tally's would prevent the Atem connection drops?

AronHetLam, 2022-03-17:
What exactly do you mean by tally server? If you are referring to the second Arduino file, then that's only acting as a fake switcher for when I'm testing things at home. The server functionality is built in by default into the tally light, so they all act as a server as well as a tally. That way you can connect them to each other, instead of all to the switcher. I'm not sure if that helps though with your WiFi. Also I'm not able to do extensive testing, other than the small two cam setup I have  inside, so I can't really say much about range. I've made the code able to run on an ESP32 as well, which should be able quite more powerful than the ESP8266, but again, I can't do extensive testing. But that said, with it working on the ESP32, it should be quite simple to use something like an m5stack ATOM https://shop.m5stack.com/products/atom-matrix-esp32-development-kit

bfbones, 2023-05-03:
Just a small update now after two productions using the tally lights: after some system changes they're working perfectly.
I now use a ESP32 build in into my rack acting as a server for those 4 tallys. As mentioned before I have a Unifi Mesh-AP which is PoE powered and I can just throw that using 50m LAN near the main event area to have perfect coverage. My cam ops had no connection drops or delays on the tallys anymore.
This brings me to that the ATEM itself (Mini Extrem Pro) has some strange problems on it's firmware unable to serv 4 connections for the tallys. That single connections works.

arkestlerdev, 2024-03-17:
@bfbones could you elaborate on how you did the “server” build please?