AronHetLam/ATEM_tally_light_with_ESP8266 #59: Tem como fazer esse sistema funcionar no M5StickC PLUS ESP32 ?

shamilton55, 2021-11-16:
I use M5StickC PLUS as tally light but unfortunately it's limited in ATEM to 4 Tally with this Tally server system I could put more tally light without limits.

AronHetLam, 2021-11-23:
Yes, theoretically it should work, however, I don't have an ESP32 myself to test with. Therefore i also don't have display support built in for the M5StickC. But take a look at pull request https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/pull/58, where @paulkilroy has proposed code that works for the ESP32. His current code is available here https://github.com/paulkilroy/ATEM_tally_light_with_ESP8266/tree/iss56.

AronHetLam, 2021-11-25:
[Release v3.1.3](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/releases/tag/v3.1.3 ) fixes the last issue to make it work on ESP32 (Which is the microcontroller M5StickC is based on. However, I see that the M5StickC doesn't expose that many pins, so you probably need to remove some LED code and replace it with display code.