AronHetLam/ATEM_tally_light_with_ESP8266 #62: Second LED with only red-green color

jmvb-cl, 2021-12-04:
It is possible add a second red LED for example, locate it in the vision sight of talent, but unlike the one which is in the camera man sight, only turns on the red (and green) lights?
I mean, no blue when connecting, no white when is no connection or no pink when is no Atem communication?
Only red and green?

jmvb-cl, 2021-12-04:
> Yes and no. The option is there to put a second led on three other pins (take a look in the guide) that you can put into program only mode, however it will still be blue, pink and white when not connected. You could solder only the red pin, but it would still be red when "pink" or "white".
> 
> Another option which would give you only red when in program is using an LED strip and selecting "no status led" in the settings, and "program only" for led 1, and put the operator led on the pins for led 2.
> 
> It should be described further in the guide.

I mean, like this guy, who is using your proyect

https://youtu.be/wSt9Ec2ssl8

AronHetLam, 2021-12-04:
I assume he's using my project, but it has to be either a very early version, or he has changed the code to not show the connection status.

jmvb-cl, 2021-12-13:
There is no way to change the current code blue-pink color behavior?

AronHetLam, 2021-12-13:
There is a way. 

If you replace `color` with `LED_OFF` in line 371 shown in below snippet, it should turn off the second LED whenever it's indicating status, as I'm using `setBothLEDs()` for that. 

https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/74231d337797b924ac395d2ae996543dfe585f84/ATEM_tally_light/ATEM_tally_light.ino#L368-L372

jmvb-cl, 2021-12-13:
371            setLED(color, PIN_RED2, PIN_GREEN2, LED_OFF);

Like this?

AronHetLam, 2021-12-13:
> 371 setLED(color, PIN_RED2, PIN_GREEN2, LED_OFF);
> 
> Like this?

No, like trhis: `setLED(LED_OFF, PIN_RED2, PIN_GREEN2, PIN_BLUE2);`

closed 2021-12-13 by AronHetLam

jmvb-cl, 2021-12-13:
Super cool! done flawlessly!
Thank you!