AronHetLam/ATEM_tally_light_with_ESP8266 #63: Failing to "Initialize" connection with ATEM

lycwolf, 2021-12-08:
First off, using and ESP32 (M5 Stick C and M5 Stick C Plus)...

I'm working on trying to port/convert this project to work natively with the M5 Stick devices, and other than the connection issues, it's looking really good.

Everything works fine except MOST of the time I will not get past the initial "Connect" part and it just waits for initialization.

The serial output:

```
M5StickC initializing...OK
त⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮HhUɥ⸮⸮⸮started
Tally1
------------------------
Connecting to WiFi...
Network name (SSID): WHPROD-WIFI
------------------------
Connected to WiFi:   WHPROD-WIFI
IP:                  192.168.1.44
Subnet Mask:         255.255.255.0
Gateway IP:          192.168.1.1
------------------------
Connecting to switcher...
Switcher IP:         192.168.1.47
Sending connect packet to ATEM switcher on IP 192.168.1.47 from port 56594
rpID: 0
rpID: 0
rpID: 1
rpID: 2
rpID: 3
rpID: 4
rpID: 5
rpID: 6
rpID: 7
rpID: 8
rpID: 12
rpID: 13 - ACK!
rpID: 14 - ACK!
rpID: 14
rpID: 14
rpID: 15 - ACK!
rpID: 15
rpID: 16 - ACK!
rpID: 16
rpID: 17 - ACK!
rpID: 17
rpID: 18 - ACK!
rpID: 18
```
What DOES work, using other/original projects made on the M5 Stick device using the original ATEMstd library instead of the modified ATEMmin library. I don't know if this has anything to do with it. I briefly attempted to port the mods over to the ATEMstd library but I'm not good enough to get it working in the short time I messed with it. Most of the other projects never use the "Initialize" function from the original ATEM libraries. But without using Wireshark or something, I can't really see the whole connection stream.

What I've tried:

- Two different physical networks. Home network and a scratch built production network. Tested both merged 5Ghz and 2.4Ghz APs and segregated 2.4Ghz APs.
- DHCP AND Static IPS on the switchers/Tally lights.
- 2 Different models of the ESP32 based M5 Stick C (Plus and non plus).
- 2 different ATEM Switchers (Atem Mini Pro Iso and Extreme Iso)
- Varies tests removing all un-needed code. Which still only works sometimes (so not completely broken).
- This included removing the LED functions (leaving NeoPixel) and adding in the screen functions for the M5 Stick (which works REALLY well when it's connected).
- Different IP ranges and subnets (basic 192.168.1.x/24 and larger 10.0.x.x/20)

The ATEM control and setup software has no issues connecting. Neither does any of the iPad control apps. Rebooting or network cycling the switchers do not affect the behavior either.

Any advice, suggestions, help, etc would be awesome! Honestly, this is one of the best projects like this out there, and the fact that you expanded the functionality of the old ATEM code is great. I've been working through the SDK from BlackMagic to add some stuff myself once I figure that whole process out (camera control would be cool too).

AronHetLam, 2021-12-08:
I assume you're using v3.1.3 😊

The "sometimes" part is strange. Are you sure you're not trying to connect to many devices to your ATEM (just making sure)? The `isRejected` flag should be set if that's the case, and it should write "Connection rejected - No empty spot" as status on the setup page.

 Is your serial output from one of the times it worked? I assume you're using `atemSwitcher.serialOutput(0x80);` to get the debug info.
By adding `#define ATEM_debug 1` at the top of the Arduino file, you should be able to get even more debug info from the ATEM lib. Just make sure to remove it before putting anything into production.

In issue #56 @paulkilroy helped getting it to work on an ESP32, so it should be possible on the m5stick-c as well.
What happens if you exchange the ATEMstd and ATEMbase libs in one of your other working projects with my modified ATEMmin and ATEMbase libs (assuming they don't need the ATEMstd features missing in ATEMmin)?

AronHetLam, 2021-12-08:
Also, could you post a link to a working project?

lycwolf, 2021-12-08:
Only a max of two devices connected at a time (including trying to connect the Tally).

Serial output is from when it doesn't work. When it does, it's ALL ACK packets. Not getting a rejected flag.

I'll try the more verbose logging.

Working projects:
https://github.com/oneguyoneblog/tally-light-esp32-for-blackmagic-atem-switcher
https://github.com/aaronpk/atem-tally-controller

I'll give the "exchanging libraries" a try when I get a chance. I'll try the verbose serial and post that back here soon (getting ready for an event so I've got a few projects going on).

Thanks for the reply!

lycwolf, 2021-12-08:
Here's the output with the other debugging on:

```
M5StickCPlus initializing...OK
`⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮HhUɥ⸮⸮⸮started
Tally2
------------------------
Connecting to WiFi...
Network name (SSID): WHPROD-WIFI
------------------------
Connected to WiFi:   WHPROD-WIFI
IP:                  192.168.1.50
Subnet Mask:         255.255.255.0
Gateway IP:          192.168.1.1
------------------------
Connecting to switcher...
Switcher IP:         192.168.1.47
Sending connect packet to ATEM switcher on IP 192.168.1.47 from port 64830
rpID: 0
rpID: 0
rpID: 1
rpID: 2
rpID: 3
rpID: 4
rpID: 5
rpID: 6
rpID: 7
rpID: 8
rpID: 12
rpID: 13 - ACK!
rpID: 14 - ACK!
rpID: 14
rpID: 14
rpID: 15 - ACK!
rpID: 15
rpID: 16 - ACK!
rpID: 16
rpID: 17 - ACK!
rpID: 17
rpID: 18 - ACK!
rpID: 18
rpID: 19 - ACK!
rpID: 19
rpID: 20 - ACK!
rpID: 20
```

Here is output from a working project:
```

M5StickCPlus initializing...OK
Sending connect packet to ATEM switcher on IP 192.168.1.47 from port 62920
Sending connect packet to ATEM switcher on IP 192.168.1.47 from port 50930
rpID: 0
rpID: 0
rpID: 1
rpID: 2
rpID: 3
rpID: 4
rpID: 5
rpID: 6
rpID: 7
rpID: 9
rpID: 13 - ACK!
rpID: 14 - ACK!
rpID: 15 - ACK!
rpID: 15
```

As for testing the ATEMmin on  the ATEMstd working project, they both call different methods for tally info and one of them has a function for remotely controlling (cutting to the camera whose tally unit it is) via a button. I want to try moving your changes into the ATEMstd library and seeing if that helps, but right now, I don't know if that's easy to do.

AronHetLam, 2021-12-09:
This is on another ESP32 than the m5stickC, with my own test server instead of an actual ATEM, but this is the debug info i get by putting my ATEMmin into "one guy one blogs" version. 

I had to comment out `Serial.begin(9600);` before `M5.begin()` wouldn't stall,  and replaced both `AtemSwitcher.getProgramTally(cameraNumber)` with `AtemSwitcher.getTallyByIndexTallyFlags(cameraNumber)`

Fuctionally it doesn't do the tally right, but I was only testing the connection.

```
M5StickC initializing...OK
Sending connect packet to ATEM switcher on IP 192.168.1.50 from port 63355
Sending connect packet to ATEM switcher on IP 192.168.1.50 from port 63154
rpID: 0, Head: 0x2, Len: 20 bytes
rpID: 0, Head: 0x0, Len: 0 bytes
rpID: 1, Head: 0x1, Len: 62 bytes
_initPayloadSent=TRUE @rpID 2
Session ID: 21419
rpID: 2, Head: 0x1, Len: 12 bytes - ACK!
ATEM _hasInitialized = TRUE
rpID: 3, Head: 0x1, Len: 12 bytes - ACK!
rpID: 4, Head: 0x1, Len: 12 bytes - ACK!
rpID: 5, Head: 0x1, Len: 62 bytes - ACK!
rpID: 6, Head: 0x1, Len: 62 bytes - ACK!
rpID: 7, Head: 0x1, Len: 62 bytes - ACK!
rpID: 8, Head: 0x1, Len: 12 bytes - ACK!
rpID: 9, Head: 0x1, Len: 12 bytes - ACK!
rpID: 10, Head: 0x1, Len: 12 bytes - ACK!
rpID: 11, Head: 0x1, Len: 12 bytes - ACK!
........
```

lycwolf, 2021-12-12:
Alrighty, I've learned a few things. This will be a little long. I've attached the debug log snippets to save people from scrolling.

First off, when I enabled the debug in the ATEMbase file, it errored out in the ATEMmin.cpp file for an undefined function. I made these changes:

Note in the original file the temp = streaming; was there, and doesn't work (for me) so I pulled the variable you defined in the ATEMmin.h file "streamingStatusFlags"

```
			/**
			 * Added by Aron N. Het Lam
			 * Functionality to parse and retrieve streaming status.
			 */
		else 
			if(!strcmp_P(cmdStr, PSTR("StRS"))) {
				#if ATEM_debug
				temp = streamingStatusFlags;
				#endif
				streamingStatusFlags = word(_packetBuffer[0], _packetBuffer[1]);
				#if ATEM_debug
				if ((_serialOutput==0x80 && streamingStatusFlags!=temp) || (_serialOutput==0x81 && !hasInitialized()))	{
					Serial.print(F("streaming = "));
					Serial.println(streamingStatusFlags);
				}
				#endif
			}
```

Next thing I noticed... Once I got the debugging info flowing properly, THE UNIT WAS ACTUALLY INITIALIZING 50% OF THE TIME! I reversed the changes to disable the debugging and didn't get it to connect again with any regularity.

So, I pulled the logging data from 3 tests. The first is what it looks like when it connects, initializes and works (screen updates with the state of the switcher. I have it change the screen color instead of using a RGB LED).
[WORKING debug log.txt](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/7697961/WORKING.debug.log.txt)

The second log shows when it just won't initialize.

[No Initialize debug log.txt](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/7697962/No.Initialize.debug.log.txt)

And the third log shows that even though it's not "connected" and the initialization never happened, I still RECEIVE the switcher status updates (but the screen never updates since it never "initialized"). 
[No Initialize debug log with switcher status updates.txt](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/7697963/No.Initialize.debug.log.with.switcher.status.updates.txt)

So, it appears that there is a connection being made, even though it's not receiving or acknowledging the initialization response. Therefore the rest of the program doesn't work because it's stuck. 

Is there an easy way to either re-send the initialization request more often (like within the connect loop) or to ignore that altogether if the device is receiving status information?

I hope this helped a bit. I may try and make some of those changes. I still want to try and get the ATEMstd working with your mods, so that I can send commands back to the switcher if needed.

lycwolf, 2021-12-12:
> Alrighty, I've learned a few things. This will be a little long. I've attached the debug log snippets to save people from scrolling.
>
> First off, when I enabled the debug in the ATEMbase file, it [...]

It also looks like the Initialization part of this is causing the units to not work when using eachother to relay the switcher data (one unit keeps trying to connect to the ATEM (the base unit in this case) and times out, while the "base" unit shows:

```
10.0.4.230:55247 - Client connected

10.0.4.230:55247 - Resent hello packet to client

10.0.4.230:55247 - Resent hello packet to client

10.0.4.230:55247 - Resent hello packet to client

10.0.4.230:55247 - Client disconnected - Was not initialized
```

AronHetLam, 2021-12-12:
From glancing over your logs I'm not sure what exactly is going on. It could seem like it's not getting all initialization packages, which makes it ask for the missing ones.
The way the lib knows (assumes) the initialization packages have all been sent, is once it gets a package that's only 12 bytes long. At that point it checks if it missed any packages up until that point, and asks for it. Only when it receives the missed packages will it set it's state to initialized.

Could you get me a log of when it's working with the one guy one blog version?

AronHetLam, 2021-12-12:
Just noticed that the `head` is often `0x5`, which means the `ATEM_headerCmd_Resend` flag is set. This could point towards there being a lot of data loss for some reason, resulting in the ESP never getting all the packages to respond to, as the ATEM is re-sending information at it's own accord.

Apparently this data loss can get the library into a state where it still needs some of the initial packages, while the ATEM thinks everything is fine. Since the lib has to parse initialization packages, it parses the state updates from the ATEM even tough it's still wanting a missed package.

lycwolf, 2021-12-13:
Neither of the other projects use the atem.Initialize command. They just do atem.connect and atem.runloop in the main loop. 

Looking through the Skaarhoj examples, it seems they only use the "initialize" routines when doing things that control the switcher remotely, as the ATEMmin and ATEMstd examples are bare-bones to just read data. Removing that (forcing it to "be connected") part of the code from this project DOES work. But now I'm wondering what's the difference in hardware with the ESP32 that is making the packet timing or data loss issues pop up. I didn't really find much info in their documentation on when/where to use the initialize command.

But this all does give me a ton more insight as to what needs done to get this all working for what I need. I'll just have to do more experimenting.

AronHetLam, 2021-12-13:
But could you still get me a log of a when it's working with one guy one blogs version? I'd like to see if it has the same behavior or not, and just works because it doesn't wait for the full initialization 😊

lycwolf, 2021-12-13:
Here we go:
[Working log from other project.txt](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/7704291/Working.log.from.other.project.txt)

AronHetLam, 2021-12-13:
Interesting, It's the same thing but they're just ignoring the fact that it is never fully initialized. I'm surprised if this is stable - is it? The tally is constantly asking for an old package, and the ATEM is re-sending all it's data... It's definitely not how it's intended to work.

lycwolf, 2021-12-13:
It seems to send all the data at the beginning, then just reads the tally data when it's actually changed:

(This is where I was pressing the CUT button on the ATEM)

Asking for package 8
rpID: 29, Head: 0x1, Len: 300 bytes - ACK!
atemTallyByIndexTallyFlags[a=0] = 1
atemTallyByIndexTallyFlags[a=1] = 2
atemProgramInputVideoSource[mE=0] = 1
atemPreviewInputVideoSource[mE=0] = 2

It's otherwise pretty dead simple (but yes, is constantly asking for the same data, and comparing the old/new state)

```
void loop() {

  // Check for packets, respond to them etc. Keeping the connection alive!
  AtemSwitcher.runLoop();

  int ProgramTally = AtemSwitcher.getProgramTally(cameraNumber);
  int PreviewTally = AtemSwitcher.getPreviewTally(cameraNumber);

  if ((ProgramTallyPrevious != ProgramTally) || (PreviewTallyPrevious != PreviewTally)) { // changed?

    if ((ProgramTally && !PreviewTally) || (ProgramTally && PreviewTally) ) { // only program, or program AND preview
      drawLabel(RED, BLACK, LOW);
    } else if (PreviewTally && !ProgramTally) { // only preview
      drawLabel(GREEN, BLACK, HIGH);
    } else if (!PreviewTally || !ProgramTally) { // neither
      drawLabel(BLACK, GRAY, HIGH);
    }

  }

  ProgramTallyPrevious = ProgramTally;
  PreviewTallyPrevious = PreviewTally;
}
```

So, in this case, as long as the session stays alive, it will intercept the tally data and update the display.

But as far as the original Skaarhoj examples go, non of the ATEMmin or ATEMstd code does the full initialization.

I wonder if there's a way to get the ESP32 to be a bit more reliable? It looks like the Skaarhoj system for ATEM actually uses a Raspberry PI via wired ethernet to convert the ATEM data (The TCP converter box they make) to something more reliable for downstream control/feedback. For more mission critical situations, that's probably the better option.

AronHetLam, 2021-12-13:
These libraries were made specifically for Arduino and in their demos they have it wired with ethernet. But yes, Skaarhoj has systems for raspberry pi as well. They stopped publishing changes to the open source Arduino libs, so this is what we got.

I'm not sure if this is an ESP32 specific problem, or if it's because ATEM firmware has changed, and the protocol is a bit different. I'd assume it would be the same on the ESP8266.

We had some issues with our ATEM video feed to OBS freezing after I updated the ATEMs firmware, so we downgraded again. It just struck me recently that it might have been freezing because of the tallys not using an updated protocol.

I didn't have these issues when I initially made the project.

AronHetLam, 2021-12-13:
But yes, the way the protocol works (worked) is by initially sending all of its state to the client, which would be the initialization, at the beginning, and thereafter jus sending new data when changes occour. If no changes happen the ATEM will keep sending packages with no new data, jus to keep the connection alive.

All of this happens in ATEMbase

If the client missed one or more of the init packages it will ask for it to be re-sent.

Maybe this behaviour has changed, which would require some reverse engineering with the latest ATEM firmware.

AronHetLam, 2021-12-26:
I peeked at how the companion module for streamdeck handles the ATEM connection, and found that they do it a bit differently.

If I get the time in January I might be able to change ATEMbase into something a bit like their approach.

lycwolf, 2021-12-27:
That's actually a great idea. I use Companion a lot, and it didn't occur to me to try checking their method. Another idea... Have a companion plugin for an ESP32 type device so that Companion is the master "hub" for the ATEM data.

AronHetLam, 2022-02-01:
Sorry for the delay.. I've been busy with exams..

It's not tested with an actual ATEM yet, but in the [refactor-conenction](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/refactor-conenction) branch you can find my initial approach of fixing it/making it mimic companions way of handling the connection.

AronHetLam, 2022-03-09:
Finally got to test it with the ATEM, and this is the output i get now:
[ESP32 Log with refactored connection.txt](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/8216406/ESP32.Log.with.refactored.connection.txt)

lycwolf, 2022-03-09:
I'll run through it and some testing this weekend. I'd say that looks better now.

AronHetLam, 2022-03-13:
That should fix the server functionality on the ESP32 :)

lycwolf, 2022-03-14:
Alright, got the M5Stick working directly to the switcher and it's working great. The only issue now is using one as a server for another M5Stick. It does a little dance to connect and this is the output:

Here's serial output from the SERVING device:
`0.0.0.0:18245 - Client disconnected - Was initialized
0.0.0.0:23635 - Client disconnected - Was initialized
rpID: 12, Head: 0x11, Len: 12 bytes - ACK!
Asking for package 7
10.0.4.122:61342 - Client connected
rpID: 13, Head: 0x11, Len: 12 bytes - ACK!
rpID: 13, Head: 0x5, Len: 12 bytes
Asking for package 7
rpID: 14, Head: 0x11, Len: 12 bytes - ACK!
rpID: 14, Head: 0x5, Len: 12 bytes
Asking for package 7
rpID: 14, Head: 0x5, Len: 12 bytes
Asking for package 7
10.0.4.122:61342 - Resent hello packet to client
rpID: 15, Head: 0x11, Len: 12 bytes - ACK!
rpID: 15, Head: 0x5, Len: 12 bytes
Asking for package 7
rpID: 16, Head: 0x11, Len: 12 bytes - ACK!
rpID: 16, Head: 0x5, Len: 12 bytes
Asking for package 7
10.0.4.122:61342 - Resent hello packet to client
rpID: 17, Head: 0x11, Len: 12 bytes - ACK!
rpID: 17, Head: 0x5, Len: 12 bytes
Asking for package 7
rpID: 18, Head: 0x11, Len: 12 bytes - ACK!
rpID: 18, Head: 0x5, Len: 12 bytes
rpID: 18, Head: 0x5, Len: 12 bytes
Asking for package 7
10.0.4.122:61342 - Resent hello packet to client
rpID: 19, Head: 0x11, Len: 12 bytes - ACK!
Asking for package 7
10.0.4.122:61342 - Client disconnected - Was not initialized
rpID: 20, Head: 0x11, Len: 12 bytes - ACK!
Asking for package 7
rpID: 21, Head: 0x11, Len: 12 bytes - ACK!
`

Here's serial output from the CLIENT device:
`Sending connect packet to ATEM switcher on IP 10.0.4.125 from port 55884
Connection to ATEM Switcher has timed out - reconnecting!
Sending connect packet to ATEM switcher on IP 10.0.4.125 from port 50803
Connection to ATEM Switcher has timed out - reconnecting!
Sending connect packet to ATEM switcher on IP 10.0.4.125 from port 64757
rpID: 0, Head: 0x2, Len: 20 bytes
rpID: 0, Head: 0x0, Len: 0 bytes
Connected to switcher
0.0.0.0:49994 - Ack not recieved - Resent tally data
0.0.0.0:173 - Ack not recieved - Resent tally data
0.0.0.0:14575 - Client disconnected - Was initialized
0.0.0.0:10761 - Client disconnected - Was initialized
0.0.0.0:49994 - Client disconnected - Was initialized
0.0.0.0:173 - Client disconnected - Was initialized
0.0.0.0:50697 - Client disconnected - Was initialized
rpID: 0, Head: 0x2, Len: 20 bytes
rpID: 0, Head: 0x0, Len: 0 bytes
rpID: 0, Head: 0x2, Len: 20 bytes
rpID: 0, Head: 0x0, Len: 0 bytes
rpID: 0, Head: 0x2, Len: 20 bytes
rpID: 0, Head: 0x0, Len: 0 bytes
Connection to ATEM Switcher has timed out - reconnecting!
Sending connect packet to ATEM switcher on IP 10.0.4.125 from port 55292
------------------------
Connection to Switcher lost...
------------------------
Connecting to switcher...
Switcher IP:         10.0.4.125
Sending connect packet to ATEM switcher on IP 10.0.4.125 from port 56456
Connection to ATEM Switcher has timed out - reconnecting!
Sending connect packet to ATEM switcher on IP 10.0.4.125 from port 59302
Connection to ATEM Switcher has timed out - reconnecting!
Sending connect packet to ATEM switcher on IP 10.0.4.125 from port 57489
`

Using DHCP and with Manual IP set for both units.

Also, it seems for some reason, the only time I get it to attempt to connect is if the SERVING device is restarted while the CLIENT device is running. That's the method I used to get the above output.

AronHetLam, 2022-03-14:
Did you use the latest version of the TallyServer library? I updated they yesterday, which should fix it.
Also yes, it's on purpose that the tally only works as servers if they are connected. This is so that the system will not give false status in case of failure.

lycwolf, 2022-03-14:
Yup, I pulled the whole zip a few hours ago.

Turning on the debugging in the TallyServer.h file gave me what mostly looks the same. The server logs just show "10.0.4.122:58657 - Resent hello packet to client"  and the client "Connection to ATEM Switcher has timed out - reconnecting!" then just sits there and resends the connection packet, trying to reconnect, which doesn't do anything until the server device is rebooted and the whole process happens again.

AronHetLam, 2022-03-14:
Hmm, it doesn't look right. You're not supposed to get this part anymore as a part of yesterdays fix. 
```
0.0.0.0:49994 - Ack not recieved - Resent tally data
0.0.0.0:173 - Ack not recieved - Resent tally data
0.0.0.0:14575 - Client disconnected - Was initialized
0.0.0.0:10761 - Client disconnected - Was initialized
0.0.0.0:49994 - Client disconnected - Was initialized
0.0.0.0:173 - Client disconnected - Was initialized
0.0.0.0:50697 - Client disconnected - Was initialized
```
Did you replace it in you `Arduino/libraries` folder?

On my ESP32 TallyServer it looks like this:

```
192.168.1.50:50157 - paket size: 20, paket length: 20
192.168.1.50:50157 - Client connected
192.168.1.50:50157 - Resent hello packet to client
192.168.1.50:50157 - paket size: 12, paket length: 12
192.168.1.50:50157 - Client Initialized
192.168.1.50:50157 - paket size: 12, paket length: 12
192.168.1.50:50157 - Ack recieved
192.168.1.50:50157 - Ack request sent
192.168.1.50:50157 - paket size: 12, paket length: 12
192.168.1.50:50157 - Ack recieved
```

lycwolf, 2022-03-14:
Yup, I made a dumb. That folder didn't get overwritten. It all works as intended now! I just need to work on M5Stick specific things like displaying the current settings, or changing the tally number, etc.

https://vimeo.com/688171071

I think this can be closed. I guess I'll fork it with my own changes for M5StickC and M5StickC Plus specific changes.

closed 2022-03-14 by lycwolf

AronHetLam, 2022-03-15:
Looks good😊