AronHetLam/ATEM_tally_light_with_ESP8266 #64: how to turn off the onboard blue led?

devokeon, 2021-12-12:
hi, so happy i can DIY the tally light for my lovely ATME,
just planing to make the box is transparent, but when in Preview will light up GREEN and on board BLUE Led,
how can i change the code to turn off on board led when in Preview mode.
thanks.

AronHetLam, 2021-12-26:
The blue onboard LED is connected to pin `D4` as I remember, but it's inverted. The blue LED is on when the pin is pulled low. As this is the same pin as the red pin for LED 2 it'll be off when the second LED is red.

 
If you change this line to use another unused pin it should be fixed.
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/master/ATEM_tally_light/ATEM_tally_light.ino#L57

If it's always on after the cange, add this at the beginning of the setup method:
 
```
 ​    ​pinMode​(D4, OUTPUT);
 ​    ​digitalWrite​(D4, 1);
```

arkestlerdev, 2024-03-12:
How do you apply this change in the newer ATEM_tally_light.cpp file, as these line numbers no longer exist in the .ino file?

arkestlerdev, 2024-03-12:
But I don't know what was in the original Line 57 that needed to be changed so I am not even sure what to search for.

AronHetLam, 2024-03-12:
I see. 
It's pin `D4` on the D1 mini. It's mapped to a constant that I named `PIN_RED2` in the code. Currently it's on line 224.
You can replace `PIN_RED2` there with another unused pin number/name.

I hope that gets you going. 🙂

arkestlerdev, 2024-03-12:
Got it, thanks for the help!