AronHetLam/ATEM_tally_light_with_ESP8266 #65: Tally with an M5 Atom

roterfan, 2022-02-06:
Hi!
Has anyone tried using an M5 Atom as a tally?
I have started it but it shows incorrect values on program and preview. Is there something in the basic library that is causing trouble?
I enclose the code if anyone wants a test and maybe has a solution.

#include <M5Atom.h>
#include <WiFi.h>
#include <SkaarhojPgmspace.h>
#include <ATEMbase.h>
#include <ATEMstd.h>

// Define the IP address of your ATEM switcher
IPAddress switcherIp(192, 168, 1, 240);

// Put your wifi SSID and password here
const char* ssid = "XXXXXXXXX";
const char* password = "YYYYYYYY";

ATEMstd AtemSwitcher;
int cameraNumber = 1;
int previewTallyPrevious = 1;
int programTallyPrevious = 1;
int cameraNumberPrevious = cameraNumber;

void setup()
{
    Serial.begin(9600);
    M5.begin(true, false, true);  //true, false, true
    
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.println("Connecting to WiFi..");
    }
    
    Serial.println("Connected to the WiFi network");
    delay(550);

    AtemSwitcher.begin(switcherIp);
    AtemSwitcher.serialOutput(0x80);
    AtemSwitcher.connect();   
}

void loop()
{ 
    AtemSwitcher.runLoop();
    int programTally = AtemSwitcher.getProgramTally(cameraNumber);
    int previewTally = AtemSwitcher.getPreviewTally(cameraNumber);
    
    // This prints 0 or 1 on all sources!?
    Serial.println("Camera program: " + String(programTally) +" Camera preview: " + String(previewTally));  
    delay(500);
    
    programTallyPrevious = programTally;
    previewTallyPrevious = previewTally;
    cameraNumberPrevious = cameraNumber;
}

AronHetLam, 2022-02-06:
You're always showing tally status for input 1, as cameraNumber is never increased in your code. If you print cameraNumber as well it might be helpful.

Putting in the delay will cause trouble, as the Atem loop needs to run continuously. It's better to put a timer in form of an if statement, that only runs eg every  500 ms.

I also see you're not using my version of the libraries. It's the smaller min version I use, so it doesn't have `getProgramTally(cameraNumber)` and `getPreviewTally(cameraNumber)`, but only `getTallyByIndexTallyFlags(cameraNumber)` which I use in my code. 1 is program, 2 is preview, and 3 means both. Also be aware that cameraNumber is zero-indexed, so cameraNumber `0` would be input `1` on the ATEM.

AronHetLam, 2022-02-06:
I get this to work. Should work with the std version of the lib as well, but you'll have to test it. 

```C++
#include <M5Atom.h>
#include <WiFi.h>
#include <SkaarhojPgmspace.h>
#include <ATEMbase.h>
#include <ATEMmin.h>

// Define the IP address of your ATEM switcher
IPAddress switcherIp(192, 168, 1, 50);

// Put your wifi SSID and password here
const char* ssid = "xxxxxxxxx";
const char* password = "yyyyyyyy";

ATEMmin AtemSwitcher;
int cameraNumber = 1;
int previewTallyPrevious = 1;
int programTallyPrevious = 1;
int cameraNumberPrevious = cameraNumber;
unsigned long long lastPrintTime = 0;

void setup()
{
    Serial.begin(9600);
    M5.begin(true, false, true);  //true, false, true
    
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.println("Connecting to WiFi..");
    }
    
    Serial.println("Connected to the WiFi network");
    delay(550);

    AtemSwitcher.begin(switcherIp);
    AtemSwitcher.serialOutput(0x80);
    AtemSwitcher.connect();
}

void loop()
{ 
    AtemSwitcher.runLoop();
    bool programTally = AtemSwitcher.getTallyByIndexTallyFlags(cameraNumber) & 0x01;
    bool previewTally = AtemSwitcher.getTallyByIndexTallyFlags(cameraNumber) & 0x02;

    if(millis() - lastPrintTime >= 500) {
        lastPrintTime = millis();
        // This prints 0 or 1 on all sources!?
        Serial.println("Camera program: " + String(programTally) +" Camera preview: " + String(previewTally));
    }
    
    programTallyPrevious = programTally;
    previewTallyPrevious = previewTally;
    cameraNumberPrevious = cameraNumber;
}
```

closed 2022-02-06 by AronHetLam