AronHetLam/ATEM_tally_light_with_ESP8266 #67: ATEM TALLY SERVER

jolysaverio, 2022-03-08:
I don't have any mistakes especially. It's just with the server solution.

I have to put the file ATEM_tally_test_server.ino in the ESP or the file ATEM_tally_light?

But if I put the ATEM_tally_test_server.ino file I can't fill in the ip of the atem because I don't see this option in the integrated web server (192.168.4.1).

How will the atem send the information to the ESP server of the chosen source?

Thanks in advance

jolysaverio, 2022-03-08:
If I have 2 ESP32 mini as a server:

Is the following configuration good?

Server 1 (4 sources)

IP: 192.168.1.150/24
IP ATEM: 192.168.1.250
Tally Light number:  ??????

Server 2 (4 sources)

IP: 192.168.1.151/24
IP ATEM: 192.168.1.250
Tally Light number:  ???????

ESP32-TALLY 1

IP: 192.168.1.152/24
IP ATEM: IP OF SERVER 1 = 192.168.1.150
Tally Light number: 1

ESP32-TALLY 2

IP: 192.168.1.153/24
IP ATEM : IP OF SERVER 1 = 192.168.1.150
Tally Light number: 2

ESP32-TALLY 3

IP: 192.168.1.154/24
IP ATEM : IP OF SERVER 1 = 192.168.1.150
Tally Light number: 3

ESP32-TALLY 4

IP: 192.168.1.155/24
IP ATEM : IP OF SERVER 1 = 192.168.1.150
Tally Light number: 4

ESP32-TALLY 5

IP: 192.168.1.156/24
IP ATEM : IP OF SERVER 2 = 192.168.1.151
Tally Light number: 5

ESP32-TALLY 6

IP: 192.168.1.157/24
IP ATEM : IP OF SERVER 2 = 192.168.1.151
Tally Light number: 6

ESP32-TALLY 7

IP: 192.168.1.158/24
IP ATEM : IP OF SERVER 2 = 192.168.1.151
Tally Light number: 7

ESP32-TALLY 8

IP: 192.168.1.159/24
IP ATEM : IP OF SERVER 2 = 192.168.1.151
Tally Light number: 8

Would the configuration be good?

What should I put in Tally Light number for the 2 servers?

Thanks in advance

AronHetLam, 2022-03-08:
You need to put the`ATEM_tally_light.ino` program on all of them, they act as both client and server. They just pass on the tally data.

The `ATEM_tally_test_server.ino` is just for having a fake "switcher" for me to develop with, or to test tally lights with if you don't have an actual switcher available while eg. building them. Hope that makes sense. You just put the IP address of the actual ATEM in the unit(s) you want to connect directly to the switcher. :)

jolysaverio, 2022-03-08:
Ok but I would not be limited in the number of tally?  Because it seemed to me that we would be limited to Max 5 per server.

jolysaverio, 2022-03-08:
What is the scheme shown for?

AronHetLam, 2022-03-08:
If you only need 8 tally lights you only need 8 ESP's. You can either connect 2 of them to the ATEM directly, and the rest to the first two

```mermaid
graph TD
    ATEM --> T1[Tally 1]
    ATEM --> T2[Tally 2]
    T1 --> T3[Tally 3]
    T1 --> T4[Tally 4]
    T1 --> T5[Tally 5]
    T2 --> T6[Tally 6]
    T2 --> T7[Tally 7]
    T2 --> T8[Tally 8]
```

or have one connected to the ATEM, and the rest looking something like this:

```mermaid
graph TD
    ATEM --> TT1[Tally 1]
    TT1 --> TT2[Tally 2]
    TT1 --> TT3[Tally 3]
    TT1 --> TT4[Tally 4]
    TT2 --> TT5[Tally 5]
    TT2 --> TT6[Tally 6]
    TT3 --> TT7[Tally 7]
    TT3 --> TT8[Tally 8]
```

Basically you can chain/connect them up as you like. I did put a limit, but in the software you could raise it. The limit is from testing with ESP8266. Most likely the ESP32 would be able to handle more than 5. However I haven't tried it.

If you're using ESP32's you might want to download the version in [this branch](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/refactor-conenction). It's a little newer than the latest release, and might fix some issues with ESP32, but again, I haven't been able to test it with a real ATEM, only my test server. 

Here's where you'd download the latest code as a .zip file if you use the link above.
[image]

jolysaverio, 2022-03-08:
Ok I understood the schematics but then in the web server on the line Tally Light Number, I must put the number of the source that will be selected on the atem?

AronHetLam, 2022-03-08:
Correct, the Tally Light Number is the source the given light should show tally status for. It will still forward all tally data for all sources.

jolysaverio, 2022-03-08:
So this is good?

Tally 1 - Server 1 (4 sources)

IP: 192.168.1.150/24
IP ATEM: 192.168.1.250
Tally Light number: 1

Tally 2 - Server 2 (4 sources)

IP: 192.168.1.151/24
IP ATEM: 192.168.1.250
Tally Light number: 2

TALLY 3

IP: 192.168.1.152/24
IP ATEM: IP OF SERVER 1 = 192.168.1.150
Tally Light number: 3

TALLY 4

IP: 192.168.1.153/24
IP ATEM : IP OF SERVER 1 = 192.168.1.150
Tally Light number: 4

TALLY 5

IP: 192.168.1.154/24
IP ATEM : IP OF SERVER 1 = 192.168.1.150
Tally Light number: 5

TALLY 6

IP: 192.168.1.155/24
IP ATEM : IP OF SERVER 1 = 192.168.1.150
Tally Light number: 6

TALLY 7

IP: 192.168.1.156/24
IP ATEM : IP OF SERVER 2 = 192.168.1.151
Tally Light number: 7

TALLY 8

IP: 192.168.1.157/24
IP ATEM : IP OF SERVER 2 = 192.168.1.151
Tally Light number: 8

jolysaverio, 2022-03-08:
ok so if I understood correctly, the 2 servers will act as a passthrough with the atem and transfer themselves the information to the ATEM.

In the end, the atem will receive data only from the servers?

Thanks

AronHetLam, 2022-03-08:
Should be good 🙂 The `(4 sources)` is incorrect, but probably because of ocpy/paste.

The servers will not send anything back to the ATEM, but only pass data on from the ATEM to other tally light connected to it. But yes, only the servers will connect to the ATEM.

jolysaverio, 2022-03-08:
yes copy / paste sorry.

Yes they will receive, I was wrong.

I'll try tomorrow, in any case many thanks for this valuable help.

I'll let you know tomorrow, thanks again for your support

Have a nice evening

AronHetLam, 2022-03-08:
No problem.
I'll close this issue, but you can still write in here if anything comes up. 🙂

closed 2022-03-08 by AronHetLam