AronHetLam/ATEM_tally_light_with_ESP8266 #71: "Settings saved successfully." is displayed, but nothing is reflected except the WiFi setting

MasakiHaraguchi, 2022-09-01:
Hello everyone.

I'm a beginner and there are some things I can't do no matter how many times I try, so I would like to know what I can do.
I am sure I have successfully installed the program on my board, but when I go to the "Tally Light setup" page, set each value and press "save changes", I get a message "Settings saved successfully." is displayed, but nothing is reflected except the WiFi setting (SSID, PW).
The LEDs are working fine.
Any ideas?

Mac OS 12.4
Arduino 1.8.19
The board I am using: 
https://ja.aliexpress.com/item/32848695833.html?spm=a2g0o.order_list.0.0.77c7585aPWEcMf&gatewayAdapt=glo2jpn
(Sorry, I can't use the original board due to a MIC problem.)
Board Manager: LOLIN(WEMOS) D1 R2 & mini, D1 mini (clone), NodeMCU 1.0

AronHetLam, 2022-09-06:
This sound a bit odd, but I also don't know your particular board. Currently mu best guess is that you've selected the wrong board in board manager.
The reason for WiFi settings being saved while the rest isn't being saved is most likely because WiFi settings are saved by the ESP library code, and the other settings are saved by my code.
For that I use Eeprom. Are you able to save data to Eeprom on your board in other arduino sketches, and if so, which upload settings and code do you use?

AronHetLam, 2022-09-06:
Found this issue which might be related: https://github.com/esp8266/Arduino/issues/6551

According to that you might need to set flash size to 2mb.

MasakiHaraguchi, 2022-09-14:
@AronHetLam 

Thank you kindly for your many comments.
As you said, there seems to be a 2MB problem.
And I see that lots of people are challenging a lot of issues.
I wish I could add a D1 WROOM-02 board as shown here, but I am a beginner and don't know how to do it.
I tried running the python file (below) in Visual studio, but there was no change in the 8266 board manager list in the Arduino IDE.

https://github.com/esp8266/Arduino/blob/c681587895c08944bcb4e84ba9ba86a24461f4a6/tools/boards.txt.py

AronHetLam, 2022-09-16:
There was an error in the script you found, but i got it working (see https://github.com/esp8266/Arduino/pull/8664#discussion_r973347916)
However, here a file containing what it generates if you only have it generate for the `d1_wroom_02`:
[boards.txt](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/9587060/boards.txt)
Make a copy of your existing boards.txt so you don't lose it.
This is where it needs to go (on windows): `"C:\Users\[USRNAME]\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\boards.txt"`

You probably need to restart the Arduino IDE.
This gives me this option, and I'm able to compile the code. But i don't have a board to try on.
[image]

MasakiHaraguchi, 2022-09-16:
wow…omg… Thank you! Thank you soooooo much!!! 
Your kindness is making me cry already.
I’ll try this!!
Thank you!!!

MasakiHaraguchi, 2022-09-18:
@AronHetLam 
YES!!! Yeeeeees!!
It completely works!!!
Perfect!!!
You saved me.
Thank you sooooo much!!
I’d like to buy beer for you…. how can I do that?

https://youtu.be/l1TD5CsWHbs

AronHetLam, 2022-09-18:
Looks good, and beer doesn't sound to bad 😉
The easiest might me PayPal: https://www.paypal.me/aronhetlam
Thanks!

closed 2022-09-18 by AronHetLam