AronHetLam/ATEM_tally_light_with_ESP8266 #73: Missing switcher messages

RobertButterworth, 2022-09-12:
Great repo, thanks.

I recently was testing out a 3 Tally light setup (ESP8266 - 5 Light WS2812B on USB power) with a BD ATEM Mini HDMI Switcher / Streamer and It seems to be missing messages resulting in the incorrect state. If I reboot the Tally Light then it grabs the correct state. When testing with the test_server everything seems fine. The Switcher and the 3 Lights are the only things on the WIFI network.

I have a fairly short deadline to resolve this so I was hoping for one of these if possible please:

A. How do I constantly poll the switcher for the current state of the Tally light (1 sec poll)?
B. Resolve the issue, either being the switcher or light (code/ESP)

Cheers.

AronHetLam, 2022-09-12:
This seems to a recurring problem, and my guess is that it's a mixture between the ESP8266 not being to powerful, and the network equipment in question (however, that seems odd if the tally lights are the only devices connected...).
This issue https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/issues/53 has the same theme.

- Maybe using the suggested "Higher bandwidth" option helps.
- Which version the this project are you using. Try and see if there is a difference between [v3.2.0](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/releases/tag/v3.2.0) and [v3.1.3](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/releases/tag/v3.1.3)
- Which version of the board manager are you using? I haven't worked with it for quite a while now, but fastLED and v3+ of the board manager had some issues. Maybe try the latest v2.x.x of the board manager.

But the reason the test server would work fine and the ATEM wouldn't is most likely due to the amount of unwanted data the ATEM sends, compared to my test_server, which only sends tally data.
An ESP32 should have more processing power. I haven't tested with an ESP32 with an actual ATEM, but it works with the test server.

RobertButterworth, 2022-09-12:
I'm currently using v3.1.3 and 3.0.2 of the board manger.
Ok will try the Higher bandwidth option and an earlier version of the board manager.

What is the best way to query the ATEM if I want to do a force sync? If I can't get this to work I was going to query every second as a stopgap.

AronHetLam, 2022-09-13:
As I remember their custom UDP protocol doesn't really allow querying data. The Atem just sends data about state change, and since it's UDP it's allowed to drop packets... There is a way to ask for lost packages in the protocol, but as you've experienced the Atem library used in the project basically ignores those.
Implementing proper handeling of lost packages in the Atem library would be ideal, but also complex.

RobertButterworth, 2022-09-13:
I did find that when I restarted the Tally Light it corrected itself to the right state.

AronHetLam, 2022-09-13:
Yes, when the connection is forst initialized you get the full state. From there on only changes are sent. Maybe you can force a reconnect to the at, but I think you'll run into issues with the Atem connection limit.

RobertButterworth, 2022-09-16:
I switched over to use high bandwidth (no features) and I don't think that it was the issue, but I will leave it on. I make a copy of the sketch and modified the code to monitor all tally lights by logging to serial the states. This monitor never misses a switch, which makes me think that the esp is sleeping/in idle? it will react when I switch to the light for the second time.

I think this issue happens ~5mins in. So I'm thinking it goes into an idle state and needs packet to wake up. Still digging.

AronHetLam, 2022-09-16:
Do you have a regular led to test with? Could easily be an issue with fastLed and the ESP8266. As i remember v.3.x.x changed something regarding interrupts, revealing issues in fastLed resulting in the Leds not being updated correctly. Try going back to board manager v. 2.x.x and see if that's better (not the version of my code)

RobertButterworth, 2022-09-17:
I thought that I would try one last thing before downgrading my board manager and I found something I think works.

Yes, FastLED is the issue, I added FastLED.delay(5); after the show it appears to fix the issue.
    
    //Show strip only on updates
    if(neopixelsUpdated) {
        FastLED.show();
        FastLED.delay(5);
        
My Repro steps is to starts the Tally lights cycle through the inputs then let it sit for 5,10,20,30 mins and check if it changes on the first switch. I think we can call this one done. Thanks for your help working this one out.

closed 2022-09-17 by RobertButterworth

AronHetLam, 2022-09-18:
Nice.
I might need to look into `fastLed.delay()`.

RobertButterworth, 2022-09-21:
Thanks for your help with this and again thanks for making this project.