AronHetLam/ATEM_tally_light_with_ESP8266 #74: Mr. Aron het Lam, I really need your advice

xx1ong, 2022-11-22:
Hello, Mr. Aron het Lam, I like your work very much, which has given me a very good experience in many application scenarios. Since I usually use the vmix coaching software, now I found a piece of code that allows me to use the same on esp8266, because I designed the LED bulb pin according to your code and the pin of vmix code is different. As a result, the colors are different. PVW is now green and PGM is blue. As my skill is very limited, I cannot solve this problem. I hope you can guide me to modify it so that this vmix code can also use your pin scheme. Thank you very much.
[VMIX code link:](https://github.com/jaycatnrdo/vMixTallyLIghtJ)

xx1ong, 2022-11-22:
`/**
  Tally light by http for Vmix

  To do:
  Make another LED flash when it is unconnected
  Makey blueLED LED turn on when it is working

  Blue will be used when input is not on live nor preview

*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#include <WiFiClient.h>

ESP8266WiFiMulti WiFiMulti; 

String currentHTML =  "nullJ";

String colorGreen = "#006400";
String colorYellow = "#ff8c00";
String colorBlue = "#1a3c75";

int redLED = 5; //D0 or GPIO 16
int yellowLED = 4;//D2 or GPIO 4
int blueLED = 0; //D3 GPIO0
int greenLED = 2;//D4 or GPIO to indicate power
//built in LED blue in WEMOS D1 - not  sure

String tallyPageDirectory = "/tallyupdate/?key=";

/*
  Location for changing settings
*******************************
*******************************
*******************************

*/

char wifiSSD[] = "bmdtally"; //change Wifi network
char wifiPassword[] = "520666520888"; //and Wifi Password

int selectCamera = 1; // and camera
String PCaddress = "http://192.168.10.137:8088"; //and address and port

/*
 ***************************
 ***************************
 ***************************

String currentKey = "6396b22d-9486-4c1a-957b-716d5faf3000";
String clientKeyAddress = "http://192.168.68.216:8088/tallyupdate/?key=ba8bca22-fa80-4302-a94b-c012bc466b14";
    //working links from home PC
*/

String currentKey = "6396b22d-9486-4c1a-957b-716d5faf3000";  // if you try to look for a page with this code you get x000000
String clientKeyAddress = "http://192.168.1.25:8088/tallyupdate/?key=ba8bca22-fa80-4302-a94b-c012bc466b14";
// Where the tally lights address should be stored, this is temporary

void setup() {

     clientKeyAddress = (PCaddress + tallyPageDirectory + currentKey);

  pinMode(redLED, OUTPUT);

[... 216 lines cut ...]
        Serial.println(tallyLinkLocation);
        currentKey = tallyPageText.substring(tallyLinkLocationStart, tallyLinkLocationEnd);
        Serial.print("current tally ");
        Serial.println(currentKey);

     clientKeyAddress = (PCaddress + tallyPageDirectory + currentKey);

        Serial.println(clientKeyAddress);

      }
    } else {
      Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }

    http.end();
  } else {
    Serial.printf("[HTTP} Unable to connect\n");

  }

  //make blueLED LED blink

  delay(200);

}`

AronHetLam, 2022-11-22:
Im not entirely Shure what the issue is, but assuming you have it working other than PGM being blue instead of red, the following should fix it.

My pin setup is:
D0	RED
D1	BLUE
D2	GREEN

So if you put in these lines in the beginning if code you posted I think it should work:
```
int redLED = D0;
int yellowLED = 4;
int blueLED = D1;
int greenLED = D2;
```

xx1ong, 2022-11-22:
> 

Mr. Aron het Lam, thanks for your busy reply, I will experiment and reply you again to confirm whether it can run, because many people still have the habit of using vmix, I also hope that Mr. Aron het Lam has considered developing the vmix tally code

xx1ong, 2022-11-22:
I'll take the original code
`int redLED = 5; //D1 or GPIO 5
int yellowLED = 4;//D2 or GPIO 4
int blueLED = 0; //D3 GPIO0
int greenLED = 2;//D4 or GPIO to indicate power`

To modify into

`int redLED = D0;
int yellowLED = 4;
int blueLED = D1;
int greenLED = D2;`

It doesn't work properly, but thank you very much for your reply. I will continue the experiment

AronHetLam, 2022-11-22:
Looking a bit more at the code you posted, something seems off here:
``` C++
if (currentHTML.substring(13, 20) == colorGreen)
          {
            digitalWrite(redLED, HIGH);
            digitalWrite(yellowLED, LOW);
             digitalWrite(blueLED, LOW);
            Serial.println("it is green" );
          }

          if (currentHTML.substring(13, 20) == colorYellow)
          {
            Serial.println("it is yellow" );
            digitalWrite(redLED, LOW);
            digitalWrite(yellowLED, HIGH);
            Serial.println("it is YELLOW" );
             digitalWrite(blueLED, LOW);

          }

          if (currentHTML.substring(13, 20) == colorBlue)
          {
            Serial.println("it is blue" );
            digitalWrite(redLED, LOW);
            digitalWrite(yellowLED, LOW);
            digitalWrite(blueLED, HIGH);

          }
```

When you print green you set redLED high, and there is no code to make it green.

AronHetLam, 2022-11-22:
Also possibly D2 could be the same as just 4, which could make green and yellow conflict in the code I sent you earlier. Make sure yellow and green are different pins

xx1ong, 2022-11-23:
> 也可能 D2 与 4 相同，这可能会导致我之前发送给您的代码中的绿色和黄色冲突。确保黄色和绿色是不同的引脚

我将去除黄色LED代码试试，因为只要保证PVW绿色和PGM红色就足够了。谢谢你

xx1ong, 2022-11-23:
> > 也可能D2与4相同，这可能会导致我之前发给您的代码中的绿色和黄色冲突。确保黄色和绿色是不同的引脚
> 
> 我将尝试删除黄色 LED 代码，因为它足以保持 PVW 绿色和 PGM 红色。谢谢

Or I have another idea. Could you please change your code pin and VMIX code pin to be the same, so that they can work together in a compatible way?

xx1ong, 2022-11-23:
> > > 可能D2与4相同，这可能会导致我之前发给您的代号中的绿色和黄色冲突。确保黄色和绿色是不同的引脚
> > 
> > 
> > 我将尝试去除黄色 LED 代码，因为它足以保持 PVW 绿色和 PGM 红色。谢谢
> 
> 或者我有别的想法。您能否将您的代码 pin 和 VMIX 代码 pin 更改为相同，以便它们可以以兼容的方式一起工作？

In today's test, I found that red connects to the D2 pin, blue connects to the D1 pin, and green does not connect to the D1 pin, allowing the VMIX code to work properly

AronHetLam, 2022-11-23:
> In today's test, I found that red is connected to D2 pin, blue is connected to D1 pin, green is not soldered, VMIX code can work normally

I'm a little confused at this point. Did you get it to work with the Vmix code?
Do you have the LED connected as my code requires it?

Else I see that the Vmix code uses yellowLED for PVW and redPIN for PGM
In that case the following should work, assuming you soldered you'r LED like my code requires it. 
```
int redLED = D0;
int yellowLED = D2; // Actually connected to green pin
int blueLED = D1;
int greenLED = D4; // You probably haven't connected this. Vmix code uses this to show it's on.
```

xx1ong, 2022-11-23:
> > 今天测试发现红色接D2脚，蓝色接D1脚，绿色没有焊，VMIX代码可以正常工作
> 
> 在这一点上我有点困惑。你让它与 Vmix 代码一起工作了吗？ 您是否按照我的代码要求连接了 LED？
> 
> 否则，我看到 Vmix 代码使用 yellowLED 表示 PVW，使用 redLED 表示 PGM 在这种情况下，以下应该有效，假设您按照我的代码要求焊接了 LED。
> 
> ```
> int redLED = D0;
> int yellowLED = D2; // Actually connected to green pin
> int blueLED = D1;
> int greenLED = D4; // You probably haven't connected this. Vmix code uses this to show it's on.
> ```

Hello, I mean by flashing in a different code, then no need to change the pin relationship.

AronHetLam, 2022-11-23:
Yes, I understand, but how is your LED connected to the ESP8266?

xx1ong, 2022-11-23:
> > 是的，我明明白了，但是你的LED是如何连接到ESP8266的？
> 
> 红色=D0、GROUNE=G、GRWWN=D2、蓝色=D1

Wire the ATEM 8266 according to your design

AronHetLam, 2022-11-23:
> > 是的，我明明白了，但是你的LED是如何连接到ESP8266的？
> 
> red=D0、GROUNE=G、GRWWN=D2、BLUE=D1

In that case this should work, as the Vmix code uses yellowLED for PVW
```
int redLED = D0;
int yellowLED = D2; // Actually connected to green pin
int blueLED = D1;
int greenLED = D4; // You probably haven't connected 
```

If it doesn't can you check what color your LED is when it prints `it is green` in the serial console? The LED should be red

And when it prints `it is yellow` the LED should be green

Some people have also had wierd issues like this if they selected the incorrect board in the board manager. Which board did you select?

xx1ong, 2022-11-25:
> 

Hello, sorry, I was just busy yesterday, and I chose D1 R2 & MINI.
I am constantly experimenting, but I haven't got the result yet. Thank you very much for your patient and professional reply. I will continue to study.

xx1ong, 2022-11-25:
[image]
[image]
[image]

Thank you very much Mr. Aron het Lam. Through your code, I have successfully flashed different codes to make ATEM and VMIX compatible, thank you again. I wish you a happy life and happy every day.

AronHetLam, 2022-11-26:
You're welcome.
Those look nice too 😊

closed 2022-11-26 by AronHetLam