AronHetLam/ATEM_tally_light_with_ESP8266 #75: How to fix the blue LED error on esp8266

congminh3366, 2022-12-15:
hi admin and everyone, i'm having trouble setting up esp8266 for tally lights project. all lights on esp after "setup tally light" are green. Can anyone tell me how to fix it please?

congminh3366, 2022-12-15:
[image]

AronHetLam, 2022-12-15:
It's a little blurry, but it looks like you soldered the LEDs incorrectly. You have it soldered to `5v`, which is wrong. You should have the LED's Ground pin soldered to `G` instead.

Besides that, i have a hard time telling if the rest of the pins are soldered correctly. You could also try to connect a LEF without soldering it, to see if it's placed correctly. Then it's easier to change.

congminh3366, 2022-12-16:
more precisely i used anode LED, i have reviewed the tally light setup process and it's true it's wrong there, now i only use green and red pins ( D2 , D0 ) What to do when the light is not working select it will not light anymore. Looking forward to your reply, thank you very much

congminh3366, 2022-12-16:
[image]
this is how i soldered the pins of the LED. Please tell me, are all 6 esp soldered like that or must be changed for each esp?

congminh3366, 2022-12-16:
https://user-images.githubusercontent.com/114283430/207993952-0cb45c2f-b10a-4d58-b3e1-646be6bd282e.mp4
when the light is not selected, it is turned yellow, so it is difficult to see the change when selecting the tally light

AronHetLam, 2022-12-16:
Ok, the wireing makes sense if you have common anode LEDs. However, the code must be changed for it to work.
The code pulls the pins high when their color should light up, but in your case it needs to be inverted, so that the pins are pulled low when their color should light up.

congminh3366, 2022-12-16:
so now if i want it to turn off the light completely when not selected, where can i fix it, can you help me edit it?

congminh3366, 2022-12-16:
can you give me the solution to fix that, thank you very much

AronHetLam, 2022-12-16:
Yes, had to find it 😊 I also don't have a common anode LED, so I can't really test it properly.

But setting the colors all happens here:
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/master/ATEM_tally_light/ATEM_tally_light.ino#L425-L472

Try changing the `0`'s to `1`'s, and change the first line from `settings.ledBrightness` to `255 - settings.ledBrightness`

congminh3366, 2022-12-16:
I can't find entries 0 and 1

congminh3366, 2022-12-16:
can you remote support ultraview to change for me?

AronHetLam, 2022-12-16:
Change each `0`(zero) to `1`(one) for each `digitalWrite(pinRed, 0);`. Eg. to `digitalWrite(pinRed, 1);`. 

Unfortunately I'm unable to remote.

congminh3366, 2022-12-16:
oh great, i tried again and it worked. really thank you very much.

congminh3366, 2022-12-16:
If there is a solution to alert the light when the signal is lost, I am sure it will be a great success that I have from you.

AronHetLam, 2023-01-15:
Sorry for the late reply.
What do you mean by `alert the light`? The LED should turn pink if it loses connection to the ATEM and white if it loses connection to WiFi.

congminh3366, 2023-12-21:
This is the result after following your instructions. How do you feel about
this product? Thank you for your interest in this product of mine. Below is
the product video link.
https://drive.google.com/file/d/152uBKgX5gSnybS2k7LHDN-5802UU6KYo/view?usp=sharing

AronHetLam, 2023-12-25:
Very nice end result! Thanks for sharing it 🙂

closed 2023-12-25 by AronHetLam