AronHetLam/ATEM_tally_light_with_ESP8266 #77: M5stack Core with W5500 PoE module

GSAL85, 2023-02-02:
Hi,  

I got the tally light working on several M5Stack modules (Core, Core2 and StickC Plus). All using WiFi to connect to the Atem.
In some situations however, the Wifi signal is unreliable, and therefore the tallies are not always working flawless. 
As a backup we wanted to make the system work on a M5Stack Core with an w5500 Ethernet / PoE module, so make it a wired system. 
However, when connected to the network via a wire, no connection to the Atem can be made.. It crashes after the atemSwitcher.begin(ip).
Does anybody have experience with this, connecting an M5Stack unit wired to an Atem??

AronHetLam, 2023-02-02:
Sadly i have no experience withe the MStack modules, but i might be able to point you in the right direction.
In the ATEMbase lib it's specified to use the `WifiUDP` lib instead of `EthernetUdp` if it's compiled for the `ESP8266` or `ESP32`
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/7053f69d69e344a52cf19e86165684ca24b4e3ab/libraries/ATEMbase/ATEMbase.h#L35-L39

GSAL85, 2023-02-03:
Hi Aron, Thanks for the reply. Indeed that was one of the first things i found and altered in the ATEMmin en ATEMbase. Even tried with the EthernetUdp2 instead of EthernetUdp. But no luck so far. It does compile, but when running it wont make connection to the Atem and starts to reboot the M5Core unit. Somehow the initialization of the Atem doesn't work properly.
If anyone else has a bright idea, i would love to hear.

AronHetLam, 2023-02-03:
Hmm.. any idea how far it gets through the initialization process? You can make it print more debug info or put some extra print statements in yourself.

AronHetLam, 2023-02-03:
Also, are you able to get UDP working over ethernet with other Arduino programs?

commputethis, 2023-08-13:
@GSAL85 I am currently using Tally Arbiter for our Tally lights with M5Sticks, but would like to reduce the complexity a little and remove the server component and use this one with the M5Sticks.  What did you have to change to get this to work with M5Sticks?

AronHetLam, 2023-08-14:
@GSAL85 How much do you know about programming arduino? Basically you need to replace/add to the code that updates the LEDs with code that updates the display or M5Stick LED instead.

I just double checked, and got it working on my ESP32 dev module (not M5Stick). There were some errors, but they are fixed in what i just pushed. I'll update the release version. Be sure to replace the libraries as well.
It should run on the M5Stick modules as well, and I'm able to compile the code for the M5Stick. You can try to upload the code to it, and see if you can access the setup page.

commputethis, 2023-08-16:
@AronHetLam I got it to compile after changing the references to WiFiUdp.h from WifiUDP.h.  Now to figure out how to change the screen color and turn on the built in red led.

commputethis, 2023-08-16:
well, I found a bit of info.  The internal red LED is pin G10 and the screen color can be set by doing something like `M5.Lcd.fillScreen(TFT_RED);`, but I am not sure where in the code to put them in the code.

AronHetLam, 2023-08-16:
Nor sure why you need to change it to WifiUdp.h, however, this is where the actual color change happens:
https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/master/ATEM_tally_light/ATEM_tally_light.ino#L369-L473 

One way is to put `M5.Lcd.fillScreen(TFT_RED);` in each `case` statement with the correct color, but it will call it twice in most cases, as that method is called once for each of the two LEDs. You could then remove the calls to it for `led2`, so the screen is only updated once each time.

Another approach is to do it like I did for the led strip. It's a little more advanced, but I think it's nicer. Essentially I've defined a list with colors (https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/master/ATEM_tally_light/ATEM_tally_light.ino#L73) that I use to lookup the correct value (https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/master/ATEM_tally_light/ATEM_tally_light.ino#L480-L491)

GSAL85, 2023-08-17:
@commputethis Sorry for the late reply. I do  agree with Aron on why you would need to change the WiFiUDP. For me that wasn't necessary.

With the M5Stack units it is convenient that they have the screens which you can change in any color you want, so a separate light / LED isn't required anymore. Using the code from Aron indeed you need to change it in the case statements. But you also need some additional lines in the setup and the loop to start and update the use of the screen. Check some of the [examples ](https://github.com/m5stack/M5StickC) of the Stick to see which ones are required. (https://github.com/m5stack/M5StickC) 

Also, there is an other [item](https://github.com/guido-visser/vMix-M5Stick-Tally-Light) on Github that might help.

commputethis, 2023-08-17:
Only thing I could figure on needing to change WiFiUDP is that I'm running
on Linux and it being case sensitive.