AronHetLam/ATEM_tally_light_with_ESP8266 #78: Wemos D1 mini PRO / Won't save basic configs in UI

154766zyc, 2023-02-17:
Hi.
I have flashed a D1 mini PRO first with 2.1.1but an issue persist.
Wifi config sticks, but all other configs like username, password, hostname, device id, led options etc... all disappear after browser refresh. Able to log-in without password after power off/on.

AronHetLam, 2023-02-17:
I'd recommend using the latest version which is currently v3.2.0. Some issues has been fixes since the version you use.

I'm not sure what you mean by username and password. The only login information it needs is WiFi credentials (name/SSID and password).
The settings page itself doesn't have a login (so anyone on the same network can access it).

By refresh the browser page do you mean you didn't hit the save button? There is no autosave if that's what you're expecting.

154766zyc, 2023-02-17:
Thanks Reply 。I used v3.2.0 on the d1 mini pro. After setting the ip address in the background, it was saved successfully, but the ip address was still not saved successfully when I saw it in the sereal monitor. How should I solve it?
![7991676618954_ pic](https://user-images.githubusercontent.com/118368225/219578897-0e7c2214-5ac9-437a-b90a-d8712f826414
[image]
.jpg)
[image]

AronHetLam, 2023-02-17:
It should all be described in the DIY guide.
For that particular error you need to install the FastLED library through the library manager.
Also, make sure you have the latest version on the libraries you get with my project.

154766zyc, 2023-02-20:
ok, thank you very much for your reply and teaching

AronHetLam, 2023-02-20:
You're welcome. 🙂
I'll close this issue as I assume you got it working.
Else you can still put a new comment.

closed 2023-02-20 by AronHetLam

154766zyc, 2023-02-25:
Hey, the problem has been solved, thank you very much for your support