AronHetLam/ATEM_tally_light_with_ESP8266 #79: WIFI Problems

ankreyes22, 2023-03-20:
I did everything the instructions said. The Light turn white and I needed to connect to the Wifi to set it but I cant see the Wifi name on my computer. Nothing at all and i dont know where to go to fix it. May be i forgot a set im not sure.

AronHetLam, 2023-03-20:
I'd need some more details to know what might be going on. Let's start with:
- What module are you using?
- What does it print in the serial monitor?
- Have you used the module before? (if so it probably has WiFi credentials already)
- Are you sure the LED is soldered correctly and whether it's common anode or cathode?
- What version of the board manager do you have?

Note the chip only supports the 2.4 GHz WiFi bands. You could try if eg. a phone can pick up the setup WiFi. 😊

WECProduction, 2023-05-02:
I'm having a similar issue except the LED won't come on at all. I thought with the size of our auditorium that the D1 mini pro would be better, but now it seems like nothing is working. It won't show the internal wifi network at all to do the initial setup. I've followed the instructions to the T. 

In the serial monitor it just repeats starting the connection and the ending with this. "Unable to connect. Serving "Tally Light setup" WiFi for configuration, while still trying to connect..."

This is a brand new board

LED is soldered correctly

Running 3.1.2 in the board manager

AronHetLam, 2023-05-02:
Try using the latest 2.x.x version of the board manager. I can't immediately tell what's going on.

WECProduction, 2023-05-02:
That didn't work. It actually wouldn't load onto the board at all. Got this error message.

Sketch uses 316660 bytes (30%) of program storage space. Maximum is 1044464 bytes.
Global variables use 36808 bytes (44%) of dynamic memory, leaving 45112 bytes for local variables. Maximum is 81920 bytes.

AronHetLam, 2023-05-03:
> Sketch uses 316660 bytes (30%) of program storage space. Maximum is 1044464 bytes.
> Global variables use 36808 bytes (44%) of dynamic memory, leaving 45112 bytes for local variables. Maximum is 81920 bytes.

That's not an error message, but just information about how much of the modules' memory the program is using.

If it didn't try to upload the program at the end you might have pressed the "verify" button instead of "upload". Could that be the case? Or did you gen other error messages as well?

WECProduction, 2023-05-03:
I definitely used the upload button. When I changed the version in the board manager to the current version, it uploaded without issue. Must have something to do with the last 2.x.x version. Even though I'm getting it to upload though, the board itself still isn't broadcasting a wifi signal. So I'm still a bit dead in the water. This is the recouping message I see in the Serial Monitor.

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00051360
~ld
����o��r��o|���l�l`c����|r�l�n��n�l`��r�l�$��########################
Serial started
��������������������������������������������������������������������
------------------------
Connecting to WiFi...
Network name (SSID): 
Unable to connect. Serving "Tally Light setup" WiFi for configuration, while still trying to connect...

WECProduction, 2023-05-03:
Aaron, could it be related to me trying to use D1 Mini Pro vs just the D1 Mini?

AronHetLam, 2023-05-03:
I don't remember if there is a specific D1 mini pro option, but maybe that could have an effect. Did you try the "All Flash" option under "Erase Flash"?

WECProduction, 2023-05-03:
Yeah, I did do that. I’m going to try and get a standard D1 mini board and try that before I go down this rabbit hole since I’m not very experienced in this type of thing. Very much appreciated on the assistance.

AronHetLam, 2023-05-03:
I also once had a D1 mini board which WiFi didn't work, so that could also be possibility.

WECProduction, 2023-05-03:
Yeah, I tried loading the code via the d1 mini pro first and also tried it the way it was written in the instructions for the other board.

AronHetLam, 2023-05-09:
I tried to build for D1 mini pro, but couldn't produce any errors. However, it won't upload to my regular D1 mini's, as it has less storage.
Sometimes on my D1 mini i had to hit upload multiple times before it wanted to upload to my board, maybe there's something there.
You could also try to hardcode your WiFi credentials into the code, and see if that works.
You'd do so by replacing the first occurrence of `WiFi.begin();` with `WiFi.begin("WiFi SSID/NAME", "WiFi PASSWORD");`, putting your WiFi credentials inside the quotation marks.
If it still doesn't connect, you might need to disable the line containing `WiFi.config(settings.tallyIP, settings.tallyGateway, settings.tallySubnetMask);`, which is a couple of lines above the `WiFi.begin();` command. You disable it by putting `//` at the beginning of the line. This is because the default values most likely make an invalid configuration. Enable and re-upload after you have configured a valid static IP configuration.