AronHetLam/ATEM_tally_light_with_ESP8266 #80: About use ws2812 for display

ericstrong, 2023-03-30:
What should I do if I want to use WS2812 for display and power it with a battery?

AronHetLam, 2023-03-31:
The Noepixel part of guide describes how you would solder the WS2812. You need to use pin D7 for the data pin, and adjust the settings according to how many LEDs you have. 
For battery use I'm not entirely sure, as I haven't done that myself. I'm sure there are plenty of guides available via Google.
Hope this points you in the right direction. 🙂

ericstrong, 2023-03-31:
I need to use 2812 as a tally indicator. However, the current control part of the LED strip only lights up the first light, and the rest of the lights don't work properly. When using other programs to control it, it works fine. Also, it seems to indicate a status instead of being used specifically as a tally indicator.

AronHetLam, 2023-04-01:
What version of the board manager do you have? Try the latest 2.x.x version, as v. 3.x.x introduced some issues for the FastLED library.
By default the first led would indicate status, and the rest would indicate tally. This can be changed to `Last` or `None` in the settings.