AronHetLam/ATEM_tally_light_with_ESP8266 #82: Neopixel not working correctly

Mikamuh, 2023-04-27:
Hi Everyone,

I am trying to get this to work using a NeoPixel-Stick. Somehow only the first LED is changing colors (not the correct ones though) but the remaining 7 LEDs are stuck.

I would love to get this to work but I don't know where to start searching for the solution.

[image]
[image]
[image]

AronHetLam, 2023-04-27:
Try downgrading the board manager to the latest 2.x.x version, as version 3.x.x and higher started causing bronlems when using the FastLED library. Hope that makes sense 🙂

Mikamuh, 2023-04-28:
Hi Aron,

thanks for the advice. I tried loading 2.1.1 onto the D1 Pro but when compiling I get faced with `Compilation error: redefinition of 'ESP8266WebServer server'`. 

Do you or anyone else have any input on how to proceed?

AronHetLam, 2023-04-28:
That's a new one. Have you changed the code in any way, or are you using other libraries than the ones specified? Also, just to make sure, what board have you selected in the board manager?

AronHetLam, 2023-04-28:
Wait, did you downgrade to my code to version 2.1.1, or the ESP8266 board manager in the Arduino IDE?
Make sure to use the latest version of my code, as it has the latest bugfixes and features, and downgrade the board manager to `2.7.4`:
[image]

AronHetLam, 2023-04-28:
I have now updated the guide so it specifies to use version the ESP8266 board manager version 2.7.4 :)

Mikamuh, 2023-04-28:
That did the Trick. Everything is working as expected now!
Thank you so much :)

closed 2023-04-28 by Mikamuh

dmitrylive, 2023-10-29:
I have a similar problem! I'm having trouble using the Arduino software. I started using the web version, it's very cool, thanks for this opportunity! I bought the boards as written in the description, D1 mini and D1 mini pro. D1 mini works correctly, I checked several boards. And on the D1 mini pro (with an external antenna) the neopixel works the same way as in the screenshots above. I tried installing different numbers of LEDs - the first one works as it should, and the rest periodically change color. frequency is unclear. I would like to use the board with an external antenna. I just can’t figure out if my board is faulty or if the mini pro has the same problem?

AronHetLam, 2023-10-30:
I made issue #94, which addresses the Arduino IDE problem.
The same [branch](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/fixArduinoIdeCompatabillity) mentioned there hopefully fixes the neopixel problem for the web installer too. It didn't use the older ESP8266 board versions, with that code it should.

@dmitrylive can I have you try it out on the D1 mini pro through the Arduino IDE to see if the Neopixels work correctly with boards manager version `2.7.4`?